Ultimaker/Cura #10752: Slicer fails - simple flat piece with 2 holes

noisewhisperer, 2021-11-06:
### Application Version

4.11.0

### Platform

Ubuntu 18.04

### Printer

Ender 3 

### Reproduction steps

Attempted to slice this piece.  Note I put the piece at z=1mm and selected tree support touching baseplate to avoid elephant's foot on holes (which I am testing to accept a slightly smaller piece. 

### Actual results

expected a gcode file :-)

### Expected results

got a popup about a slicer failure and suggestion to open bug report, so here I am.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

none.

noisewhisperer, 2021-11-06:
[cura.log](https://github.com/Ultimaker/Cura/files/7490167/cura.log)
[CE3_bp_holetest-01.3mf.gz](https://github.com/Ultimaker/Cura/files/7490174/CE3_bp_holetest-01.3mf.gz)

noisewhisperer, 2021-11-06:
For fun I pushed the slice button again, and slice process completed. !? good?

fvrmr, 2021-11-08:
Hi @noisewhisperer thank you for your bug report.
This is a duplicate of #10394 and is probably happening because your model is lower than the initial layer height. This is fixed in 4.12 Beta. You can download the beta here: https://github.com/Ultimaker/Cura/releases/tag/4.12-beta

I will close this issue because it is a duplicate.

closed 2021-11-08 by fvrmr