Ultimaker/Cura #10753: Disconnected brim lines

mistic100, 2021-11-06:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

When printing multiple close models with a brim it happens some brim lines are started in an empty space instead of the existing brim, whereas it will be eventually connected.

Here I am printing 6x "pink_face" for this object https://www.thingiverse.com/thing:173570/files with default automatic placement and a 3mm brim.

### Actual results

You can clearly see what I mean here, the brim is printed from inside to outside
[image]

It does not happen everytime, for exemple the brim for the model on the bottom of the screen was printed outside to inside.

This is particularly problematic with small models, the inside line having sharper corners thus more adhesion issues. It kind of defeat the purpose of brims.

### Expected results

The brim should always be printed outside to inside to avoid disconnected lines.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

project file : [CE3_blue_face_aperture.zip](https://github.com/Ultimaker/Cura/files/7490554/CE3_blue_face_aperture.zip)

fvrmr, 2021-11-08:
Hi @mistic100 thank you for your report. 
I will bring this up with the team. Keep you posted!

fvrmr, 2021-11-12:
I have discussed it with the team. And I created a ticket on our backlog for this. 
Devs see CURA-8695

mistic100, 2021-12-11:
@fvrmr Another very weird example : the brim lines are printed alternatively

[image]

Project : [CE3_PoteauAngle.zip](https://github.com/Ultimaker/Cura/files/7696773/CE3_PoteauAngle.zip)

GregValiant, 2024-11-16:
Is this still a problem in current versions of Cura (5.8.0 and up)?  Can this be closed?

mistic100, 2024-11-16:
Not really. I just sliced the file in Cura 5.8.1 still with default 0.2mm profile + 3mm brim and some lines are still disconnected.

[image]

GregValiant, 2024-11-16:
I sliced your "Poteau Angle" project with 5.8.1 and the brim is all connected.
[image]

Using 5.9.0 beta 2 the "Blue Face Aperture" project also looks correct.
[image]

This is the Blue Face Aperture project in Cura 4.13.1 (I don't have 4.11).
[image]

I used the files from the original post and cannot seem to duplicate the problem.

mistic100, 2024-11-16:
If you visualize the whole layer yes of course the lines are connected, the problem is **during** the print of the layer, if you use the bottom scrollbar you will probably see something similar to my screenshot, where some lines are printed sequentially then sometimes a line is skipped and printed later.

This cause some lines to be printed without touching the others

GregValiant, 2024-11-16:
I see that motion.
You have your "Skirt Brim Flow" set to 90%.  Setting the "Initial Layer Flow" to 100% means the brim will go down at 100% of 90%.
Each extrusion of the brim will be 0.36 wide (because the layer height doesn't change) and the nozzle index distance is 0.4 (your line width).  That will result in the brim being "thready" (0.04 gaps between the brim extrusions) instead of a solid mat.  That situation is independent of the toolpath of the brim.

This appears to be a settings issue.  If you have calibrated your E-steps then your flows should be fine at 100% which would result in actual extrusion widths of 0.4 which will match the index distance of 0.4 and the brim would be all welded together regardless of the toolpath.

If you calibrated the E-steps and then "calibrated the flow" to a "Single wall calibration cube" or something similar, then all models except for single wall models, will be under-extruded by 10%.
If you wish to leave your flow rates at 90% then you should adjust the "Skirt Brim Flow" to 100%.

I don't see a bug here but rather a setting issue and I'm inclined to close it.  The original report was for Cura 4.11.  The brim code in Cura has been reworked more than once since then and this is the only report of this behavior.
I can duplicate this behavior if I drop the Skirt Brim Flow to 90%, but at 100% the brim is a solid mat.

What are your thoughts?

mistic100, 2024-11-16:
I admit I don't really understand your last message. I don't see how the calibration would influence the toolpath. I tried with brim flow from 80 to 120% and the toolpath is always the same, with disconnected lines.

As I said this is with the **default profile and the default settings**.

https://github.com/user-attachments/assets/9ef5e69a-f9af-49e2-9b0e-59b8f92299a6

Disconnected brim lines at 0:15 for both bottom and top pieces.

Feel free to close the issue if the dev team don't think it is worth investigating. As a matter of fact whereas I consider the toolpath to be wrong, I don't have a adhesion issues since switching to a PEI bed.

mistic100, 2024-11-16:
By the way there is a similar issue here https://github.com/Ultimaker/Cura/issues/11936

GregValiant, 2024-11-16:
I think the brim toolpath should maintain concentricity as well.
The "Blue face aperture" file is a bit different because it is multiple models and the brims interfere.  Moving the models apart so that the brims don't interfere does not significantly change the toolpath.  After finishing all except for 2 loops of the brim -  a loop starts at the brim/model interface and then moves outward to complete the brim.  That may well be "expected behavior" but I don't know that.  I can see where it might affect "initial layer adhesion" for some printers.
At any rate I'll leave this open since there might be something going on there.
[image]