Ultimaker/Cura #10754: Spiralize adds extra inner walls on alternating bottom layers

UnfoldAntwerp, 2021-11-06:
### Application Version

3.1.0, 4.9.1, 4.11

### Platform

MacOS 10.15.7, 11.5

### Printer

Different printers

### Reproduction steps

1. turn on spiralize
2. 5 extra inner walls (green) are added to each alternating even bottom layer.

went through all the settings but could not find anyone causing this behaviour...

### Actual results

5 extra inner walls (green) are added to each alternating even bottom layer.

### Expected results

NO extra inner walls (green) are added to each alternating even bottom layer

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[cupIssue.3mf.zip](https://github.com/Ultimaker/Cura/files/7490587/cupIssue.3mf.zip)
[image]

fvrmr, 2021-11-08:
Hi @UnfoldAntwerp thank you for your report. 
This is a duplicate of https://github.com/Ultimaker/Cura/issues/8393
There you can read that this is hard-coded to make the model water tight. 

I will close this issue since it is a duplicate.

closed 2021-11-08 by fvrmr

UnfoldAntwerp, 2021-11-08:
Thanks @fvrmr! I did a proper search but did't surface under spiral issues. So thanks for pointing this out. Baffles me that this is on purpose but I'll respond in the older issue.