Ultimaker/Cura #10756: Arrange all parts does not work

onthuhlist, 2021-11-06:
### Application Version

4.9.0 then 4.12.0 beta

### Platform

Mac

### Printer

CR10s Pro

### Reproduction steps

I loaded 4 small simple parts at the same time. Then I quit the app and tried loading them one at a  time.

### Actual results

In all instances, all models load directly on top of one another. No way to auto-arrange them. The auto-arrange command does nothing.

### Expected results

Cura used to arrange them on the bed automatically for printing. 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

Servus7, 2021-11-07:
Got the same issue after updating to Mac OS Monterey.

fieldOfView, 2021-11-07:
Duplicate of #10133

fieldOfView, 2021-11-07:
Probably the same underlying issue as #10249

closed 2021-11-07 by nallath