Ultimaker/Cura #10757: Slicing a lithophane

AlaskainFlight, 2021-11-07:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Creality 5 pro

### Reproduction steps

Loaded a jpg photo and sliced it.

[image]
[image]

### Actual results

The preview shows that Cura omitted the base.

[image]

### Expected results

The base, the air gap under the printed photo should have been filled in with multiple layers of filament
[image]
.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
[Buddy 000.zip](https://github.com/Ultimaker/Cura/files/7491892/Buddy.000.zip)

AlaskainFlight, 2021-11-07:
Resolved. Delayed Intelligence
[image]
!

closed 2021-11-07 by AlaskainFlight

Ghostkeeper, 2021-11-15:
The problem was, as far as I can see, rather with Surface Mode. Setting that to Normal works.