Ultimaker/Cura #10760: Extruder 2 turns off the hotend early when the second material is no longer needed

Juusoz, 2021-11-07:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Custom FFF Printer

### Reproduction steps

1. Have a dual-extrusion printer with a single nozzle
2. Slice a model with two different materials
3. Save the .gcode

### Actual results

When one of the extruders has finished its job, it will turn off the hotend, as expected with multi-nozzle printers. But I have a single-nozzle printer with two extruders, so it will just turn off the only hotend the printer has mid-print, leading to cold-extrusion.

[image]

It will also initiate standby temperature in the middle of the printing, possibly dropping the temperature to below-melting point.

### Expected results

The hotend should stay warm until the end.

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

This could all be avoided if Cura had a checkbox in machine settings for single-nozzle printers.

**Log, project and .gcode files:** [SingleNozzle.zip](https://github.com/Ultimaker/Cura/files/7492702/SingleNozzle.zip)
I have an add-on in Cura that might screw with the log file a bit, but the bug can be replicated in standard version as well.

Juusoz, 2021-11-07:
And for anyone looking for a fix for this, you can add a post processing script _**Search and Replace**_ for removing the lines that turns off the hotend. For the Search add "**M104 T0 S0**" (Turn off Hotend 0) and leave the Replace line blank.
Then make another one for "**M104 T1 S0**" (Turn off Hotend 1).

The post processing script does not touch the end gcode, so no need to worry about the printer not being able to turn off.

[image]

GregValiant, 2021-11-08:
The plugin "Printer Settings" (available in the Marketplace) has numerous settings regarding the printer.  In particular are "Extruders Share Heater" and "Extruders Share Nozzle".  With those enabled your problem should go away.
[image]
e enabled you shouldn't have the problem.  They are for multiple-in-one-out hot ends.

Juusoz, 2021-11-08:
Seems like a useful plugin, will have to check it out at some point. :)

But still, I should not need to install plugins to fix a major oversight in Cura.

GregValiant, 2021-11-08:
Cura exists in a world of plugins.  Those settings were in Machine Settings but were moved to "Printer Settings".  Now the bulk of  the the machine related settings are together. Since I loaded Printer Settings once, now it always loads when I update Cura.

Ghostkeeper, 2021-11-15:
> But still, I should not need to install plugins to fix a major oversight in Cura.

Indeed, and you wouldn't need to if the printer definition you're using sets this correctly. However you're using Custom FFF Printer, for printers that don't have their own profiles in Cura. Unfortunately you'll have to deal with this sort of issue then. The same goes for any of the other machine settings save for the select few that have a place in the Machine Settings dialogue.

The Extruders Share Heater setting is the intended solution here. As far as I can tell, that works correctly.

closed 2021-11-15 by Ghostkeeper