Ultimaker/Cura #10762: App icon on macOS is too large

CraigSiemens, 2021-11-07:
### Application Version

4.11.0

### Platform

macOS

### Printer

n/a

### Reproduction steps

Launch the app on macOS.

### Actual results

The icon size is large than the other icons on macOS.

[image]

### Expected results

The icon should have similar sizing to the other icons. There are [templates](https://developer.apple.com/design/resources/#macos-apps) available to get the correct sizing.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[Apple - Human Interface Guidelines - macOS - App Icons](https://developer.apple.com/design/human-interface-guidelines/macos/icons-and-images/app-icon/)

neofuture, 2021-11-08:
Second this is goes against apples ecosystem and human interface guidelines

fvrmr, 2021-11-08:
Hi @CraigSiemens thank you for your report.
I have discussed it with our UX designer and we have created a ticket on our backlog.
Devs see CURA-8681

fvrmr, 2022-05-17:
With the new Cura 5.0 release the Cura icon has had an update and is now better aligned on MacOS. So I will close this issue!

closed 2022-05-17 by fvrmr