Ultimaker/Cura #10765: Raft does nor generate on smaller parts under the wall size

neofuture, 2021-11-08:
### Application Version

4.11.9

### Platform

macOS M1

### Printer

CR10/Ender3

### Reproduction steps

Using models that are very small  (less than 5 mm as tested, may be treated) are excluded from the raft and a raft is not generated and the raft is ignored and left out of the print, leaving prints to fail

Add a small part around 3 mm in diameter, duplicate this 50 times then add a new object 20mm and duplicate this 20 times, the 20 mm Object gets a raft the 3mm object does not

### Actual results

Rafts do not get calculated for tiny parts

### Expected results

Rafts !!!

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

Add a small part around 3 mm in diameter, duplicate this 50 times then add a new object 20mm and duplicate this 20 times, the 20 mm Object gets a raft the 3mm object does not

fvrmr, 2021-11-08:
Hi @neofuture thank you for your report.
Could you share a project file? I could not reproduce this with your reproduction steps.

neofuture, 2021-11-08:
[CE3V2_TUMBLER.3 Brokenmf.3mf.zip](https://github.com/Ultimaker/Cura/files/7499345/CE3V2_TUMBLER.3.Brokenmf.3mf.zip)

Slicing this only results in a raft on the later tumblers and the small ones do not get a raft

fvrmr, 2021-11-11:
Hi @neofuture thank you for your project file.
So looking at your models. The first layers of those small models are not printed. If you enable thin walls, they are printed and also on the raft. 

[image]

The best way to print this is using our Arachne build since that will have variable line widths. See here more about Arachne: https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta 

[image]

Let me know if that works for you!

closed 2021-11-25 by no-response[bot]