Ultimaker/Cura #10768: slicing failed

Heagr, 2021-11-08:
### Application Version

4.11.0

### Platform

windows 10

### Printer

anet a6

### Reproduction steps

trying to print a miniatur
set top layer to 9999999999
Bottom layer to 0
speed 25
100% infille
0.1 profile fine

### Actual results

nothing

### Expected results

get the slicing failed error

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

fvrmr, 2021-11-09:
Hi @Heagr thank you for your report.
Could you share a project file? (file - save project)

Ghostkeeper, 2021-11-15:
Possibly this is a problem because 9999999999 is more than the maximum integer (2^31). Could be looking at an integer overflow here.

closed 2021-11-23 by no-response[bot]