Ultimaker/Cura #10769: Modify settings per Overlap, wall count not working

zlice, 2021-11-08:
### Application Version

4.11.0

### Platform

linux

### Printer

ANYCubic i3 mega-s

### Reproduction steps

- Load part
- Create Support blocker
- Change block to "modify settings for overlaps"
- Change `wall line count` from `3` to `33`

### Actual results

Slice is same as no per-model settings

### Expected results

Wall count draws squares instead of zig-zags on overlap.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[scroll_.stl.gz](https://github.com/Ultimaker/Cura/files/7499339/scroll_.stl.gz)

nallath, 2021-11-09:
Could you add a project file that reproduces this? I've tried reproducing this myself and wasn't able to.

[image]

zlice, 2021-11-09:
I see you used the `cutting mesh` instead of `infill`. That works. Thanks!

closed 2021-11-09 by zlice

Ghostkeeper, 2021-11-15:
Indeed, the infill mesh only affects the infill. Any walls will be added inside the infill area.