Ultimaker/Cura #10770: Install Pulgins

NunoFilipePereira, 2021-11-08:
### Application Version

4.12.0

### Platform

mac

### Printer

Biqu b1 /Tarantola Pro

### Reproduction steps

Does not let you install Pulgins. When trying to install or update the ultimaker cura closes

### Actual results

Does not let you install or update

### Expected results

should let install or update, as the ultimaker cura

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I don't have much more to say. I uninstalled, re-installed and the problem continues

fvrmr, 2021-11-09:
Hi @NunoFilipePereira thank you for your report.
Could you share your log file? You can find your log file here: `$HOME/Library/Application Support/cura/<Cura version>/cura.log`

NunoFilipePereira, 2021-11-09:
Good afternoon.
At this address there is no Cura folder..

No dia 09/11/2021, às 13:02, Fenne ***@***.******@***.***>> escreveu:

$HOME/Library/Application Support/cura/<Cura version>/cura.log

fieldOfView, 2021-11-09:
In Cura, go to Help -> Show configuration folder... The cura.log file can be found in the folder that opens.

Please note that you can not send the cura.log file as an attachment in an email; you have to attach the log file via https://github.com/Ultimaker/Cura/issues/10770

NunoFilipePereira, 2021-11-09:
I've already found it. But it wasn't at the address you said. Yet here is the log. Anything else you need I'm available for the evolution of the program. Thanks

No dia 09/11/2021, às 13:59, Nuno Filipe Pereira ***@***.******@***.***>> escreveu:

Good afternoon.
At this address there is no Cura folder..

No dia 09/11/2021, às 13:02, Fenne ***@***.******@***.***>> escreveu:

$HOME/Library/Application Support/cura/<Cura version>/cura.log

fieldOfView, 2021-11-09:
you can not send the cura.log file as an attachment in an email; you have to attach the log file via the web interface at https://github.com/Ultimaker/Cura/issues/10770

NunoFilipePereira, 2021-11-09:
Ok. I already sent the file through guithub. It is impossible to work with the program. As soon as I try to synchronize the program closes. does not let me install or update pulgins and account. Please help me

nallath, 2021-11-10:
> Ok. I already sent the file through guithub. It is impossible to work with the program. As soon as I try to synchronize the program closes. does not let me install or update pulgins and account. Please help me

You need to go to the webpage and upload the logs there. We do not have the logs that you sent via email. We can't help you without them.

NunoFilipePereira, 2021-11-10:
I've already sent it through guithub

Enviado do meu iPhone
Nuno Filipe  Pereira

No dia 10/11/2021, às 08:55, Jaime van Kessel ***@***.***> escreveu:

Ok. I already sent the file through guithub. It is impossible to work with the program. As soon as I try to synchronize the program closes. does not let me install or update pulgins and account. Please help me

You need to go to the webpage and upload the logs there. We do not have the logs that you sent via email. We can't help you without them.

nallath, 2021-11-10:
> I've already sent it through guithub 

Sorry, we still don't have it.

NunoFilipePereira, 2021-11-10:
My log

Help me plesa

nallath, 2021-11-11:
> My log Help me plesa

We still don't have the logs. You can't send us logs via email. Re-sending it will not work.

Please follow these instructions: 
1. Go to https://github.com/Ultimaker/Cura/issues/10770. 
2. Add your log file via the website.

fieldOfView, 2021-11-11:
Alternatively, you can send your log file to aldo[at]fieldofview.com, and I will post it here.

NunoFilipePereira, 2021-11-11:
Hello
I have already added the log file.
Can you help me?

https://github.com/Ultimaker/Cura/issues/10770

fieldOfView, 2021-11-11:
No, we cannot help you. The way you are trying to send the log files to us does not work. They do not arrive with us. 

Please send the log file to my personal email address, aldo[at]fieldofview.com and I will make sure they are posted in the correct location, here on the github website. You will have to replace the [at] in my email address with an @.

fvrmr, 2021-11-11:
What are exactly your reproduction steps to install the plugins? Do you use the marketplace?

fieldOfView, 2021-11-12:
This is a log provided by @NunoFilipePereira in #10807: https://github.com/Ultimaker/Cura/files/7523528/cura.log

Ghostkeeper, 2021-11-15:
There is a log file in the top post here, too. I think people overlooked it because it was edited in later.

Ghostkeeper, 2021-11-15:
In the log file I see several hard crashes, where Cura crashes before even being able to log the error or show an error dialogue. However the crash always happens shortly after this log entry:

    2021-11-09 16:59:42,100 - DEBUG - [MainThread] Toolbox.src.CloudSync.CloudPackageChecker._handleCompatibilityData [127]: Discrepancy found between Cloud subscribed packages and Cura installed packages

That's when Cura will show a pop-up message that prompts the user to update them. So that seems to be how this issue is reproduced: Add a few plug-ins to your Ultimaker account, but remove them from Cura (or never install them). Then start Cura, and accept the prompt to install the packages.

When I try that to reproduce the issue, it works fine. However I am on Linux and Nuno is on MacOS. Perhaps the issue only occurs on MacOS. Or perhaps the issue is only with select plug-ins that don't function for MacOS (like the CAD importer?). Or maybe Cura can't deal with a spotty internet connection well in this case; I do see some log entries about that too earlier on.

nallath, 2021-11-15:
> In the log file I see several hard crashes, where Cura crashes before even being able to log the error or show an error dialogue. However the crash always happens shortly after this log entry:
>
> ``` [...]

That's because he has installed plugins from source (as you can see from the -master prefix of the plugin id's).

Ghostkeeper, 2021-11-16:
So it's crashing when trying to update a plug-in that was side-loaded?

GregValiant, 2024-11-16:
Is this still a problem in current versions of Cura (5.8.0 and up)?  Can this be closed?

closed 2024-12-09 by github-actions[bot]