Ultimaker/Cura #10775: Closes when updating account

NunoFilipePereira, 2021-11-09:
### Application Version

4.12.0

### Platform

Mac

### Printer

Biqu b1 /Tarantola Pro

### Reproduction steps

When I try to synchronize the account to update pulgins and other data, the program closes.

### Actual results

Already uninstalled and re-installed and the problem continues. Appeared since I installed the beta version of 4.12. It is therefore impossible to work with the program without some plugins

### Expected results

I await a resolution as soon as possible

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7506755/cura.log)

fvrmr, 2021-11-11:
Hi @NunoFilipePereira thank you for your report. 
I could not find anything in the logs that shows why this is happening. 
What version OS are you using?

Z122-JPL, 2021-11-14:
@NunoFilipePereira 
Here is the solution to fix this known 'Sign In" problem under Mac OS:
https://github.com/Ultimaker/Cura/issues/10789#issuecomment-968054449

NunoFilipePereira, 2021-11-15:
> Hi @NunoFilipePereira thank you for your report.
> I could not find anything in the logs that shows why this is happening.
> What version OS are you using?

MacOs Catalina Versão 10.15.7

fvrmr, 2021-11-16:
@NunoFilipePereira thank you
Could you check the comment @Z122-JPL mentioned if that works for you: https://github.com/Ultimaker/Cura/issues/10789#issuecomment-968054449

closed 2021-11-30 by no-response[bot]