Ultimaker/Cura #10776: "Multiply" doesn't work on M1_ processors.

jonathanalevi, 2021-11-09:
### Application Version

4.12

### Platform

Mac OS

### Printer

Geeetech A10M

### Reproduction steps

Tried to multiply a basic shape.

### Actual results

Stuck on "Placing Objects" screen. 

### Expected results

Objects should be multiplied. 

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

<img width="473" alt="image" src="https://user-images.githubusercontent.com/1540530/140975538-21fe5552-75d5-4aed-a374-d7c8219ee3e0.png"
[Archive.zip](https://github.com/Ultimaker/Cura/files/7506861/Archive.zip)
>

etjensen, 2021-11-09:
I have the same issue (M1 Mac Mini). I think a related issue is that when I drag/drop STLs from the Finder, they all are located at bed center, so overlap on the bed until they are moved. Earlier Cura versions would arrange them so they didn't overlap.

nallath, 2021-11-10:
Duplicate of #10249

closed 2021-11-10 by nallath