Ultimaker/Cura #10777: Top layer get's bits of infill when infill overlap is > 0%

mastercartman, 2021-11-09:
### Application Version

4.12.0-beta

### Platform

Windows 11

### Printer

Custom

### Reproduction steps

Slice part with some top layers, not matterhow many, profile is irrelevant, result is all the same. Infill Overlap is left default to 10%

### Actual results

in top layer around inner walls tiny bits of infill are placed, where the infill was some layers beneath, so printer is extensively retracting (if enabled), and traveling through the top layer. See 
[image]
the small orange bits in my screenshot.

### Expected results

As it is a top layer or skin type wall no infill should be place
[image]
d.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[z_top_motor_mnt v5.zip](https://github.com/Ultimaker/Cura/files/7507305/z_top_motor_mnt.v5.zip)

nallath, 2021-11-10:
Did you also check this for the 4.12 release? It should be fixed in between the beta and final release.

mastercartman, 2021-11-11:
Hey there, just installed the release and checked it. The issue is not present anymore, thanks!

closed 2021-11-11 by mastercartman