Ultimaker/Cura #10779: Missing Layer View in Preview

drclong, 2021-11-10:
### Application Version

4.12

### Platform

Windows 10

### Printer

Artillery Genius

### Reproduction steps

Loaded model and sliced
Select layer view but there is no analysis of the layers. Only X-Ray View is available
Searched all over for any information to enable layer view to no avail.

### Actual results

Nothing. Layer view is not available. Only  X-ray view is

### Expected results

Ability to select layer view

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[Logs.zip](https://github.com/Ultimaker/Cura/files/7508809/Logs.zip)

fieldOfView, 2021-11-10:
You seem to have inadvertently disabled the Simulation View plugin.

Go to the "Installed" tab on the Marketplace, and scroll down to Simulation View. Make sure the checkbox in front of it is checked. Then restart Cura.

drclong, 2021-11-10:
I just knew it was a simple fix. Thankyou so much. Searched everywhere for an answer to this.

closed 2021-11-10 by drclong