Ultimaker/Cura #10780: Cura 4.12.0 crashes when selecting an file with stl extension

JBatCCA, 2021-11-10:
### Application Version

4.12.0

### Platform

Windows 7

### Printer

2 extended plus and mark2

### Reproduction steps

1. Cura opens fine. I click on the folder icon to open a file. (or select "File - Open File) Navigate to anywhere on my computer with files. Select ANY file with a .stl extension. When you select a file, you normally have to "double-click" it to open it. As soon as I click it the FIRST time, everything disappears. i.e. Cura shuts down immediately like it was never open.
2. If I navigate through Windows Explorer to find any .stl file and double click on it to open it, (I have Cura set as a default program to use to open stl files) the file will open and I will be able to work with it normally.

### Actual results

Cura shuts down immediately. 

### Expected results

Cura should stay open and running.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7508960/cura.log)

nallath, 2021-11-10:
Do you have chitubox installed? If so, please check https://support.ultimaker.com/hc/en-us/articles/4402996140946-Ultimaker-Cura-crashes-when-opening-any-file

JBatCCA, 2021-11-10:
I uninstalled Chitubox and it appears to be working now. Thank you.
The removal of the thumbnail preview did not work, but I'm sure I messed that up some how. My location was specific and I wasn't sure how to use the brackets and quotes in the string, since it wasn't very clear.

closed 2021-11-10 by JBatCCA