Ultimaker/Cura #10782: Problem with Printer profiles

CrazyJason14, 2021-11-10:
### Application Version

4.12.0

### Platform

Windows 10

### Printer

that's the issue

### Reproduction steps

Upgraded to Cura 4.12 from 4.11

### Actual results

When I installed cura 4.12 after havin version 4.11, I got a Configuration error message upon opening that stated my configuration files were corrupted. Says something is wrong with the following profiles:
 - x-plus_pla_fine
 - X-Plus
 - qidi_extruder_0#2
 - x-plus_fine
 It askes if I want to reset to factory defaults which would remove my other 6 printer profiles. I got the Qidi profile from Github and loaded it into Cura 4.11 and it works perfectly, however it will not work in 4.12. When I open 4.11, my Qidi profile is present and working as expected, but I get the error message upon opening 4.12 every time and my Qidi printer is not listed in my selectable printers.

### Expected results

Either inform me of a way to get my previous profile to work in 4.12, or do a bug fix that will allow it to work properly

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No additional files needed. can send a screenshot if needed

CrazyJason14, 2021-11-10:
Here is a screenshot of the error I get when opening version 4.12
[image]

nallath, 2021-11-10:
It seems that the issue template has not been filled in entirely. Please check our guide on https://github.com/Ultimaker/Cura/wiki/Reporting-issues for more information on what information we need and why we need it.

CrazyJason14, 2021-11-10:
not sure what you are missing from the template, all pertinent information is there

nallath, 2021-11-10:
No, it's not ;) 

We ask for a log file which you didn't include.

CrazyJason14, 2021-11-10:
Here is the log file
[cura.log](https://github.com/Ultimaker/Cura/files/7516626/cura.log)

Ghostkeeper, 2021-11-17:
According to the log file, it doesn't have a profile with the filename `x-plus_fine.inst.cfg`:

    When trying to deserialize X-Plus, we received an unknown container ID (x-plus_fine)

Now it could be that it exists, but you forgot to upgrade the version of these profiles and this is just the first one it encounters.

Cura has a version upgrade system that upgrades anything automatically that the user can create through Cura. But this upgrade system is unable to automatically upgrade mods as well, such as custom plug-ins or side-loaded profiles like yours. I think you need to manually apply the changes done by [the version upgrade between 4.11 and 4.12](https://github.com/Ultimaker/Cura/blob/master/plugins/VersionUpgrade/VersionUpgrade411to412/VersionUpgrade411to412.py), which is to update the setting_version and to change references to FlSun profiles if any.

closed 2021-12-01 by no-response[bot]