Ultimaker/Cura #10785: Support Blocker crash

karl-nilsson, 2021-11-10:
### Application Version

4.11, 4.12

### Platform

Void Linux

### Printer

N/A

### Reproduction steps

1. Load model
2. Select model
3. Click support blocker button
4. Click part

### Actual results

Crash dialog pops up.
Relevant log section:
`2021-11-02 08:44:19,193 - CRITICAL - [MainThread] cura.CrashHandler.init [69]: File "/usr/lib/cura/plugins/SupportEraser/SupportEraser.py", line 91, in event
2021-11-02 08:44:19,193 - CRITICAL - [MainThread] cura.CrashHandler.init [69]: picked_position = picking_pass.getPickedPosition(event.x, event.y)
2021-11-02 08:44:19,193 - CRITICAL - [MainThread] cura.CrashHandler.init [69]: File "/usr/lib/python3.10/site-packages/cura/PickingPass.py", line 93, in getPickedPosition
2021-11-02 08:44:19,193 - CRITICAL - [MainThread] cura.CrashHandler.init [69]: distance = self.getPickedDepth(x, y)
2021-11-02 08:44:19,193 - CRITICAL - [MainThread] cura.CrashHandler.init [69]: File "/usr/lib/python3.10/site-packages/cura/PickingPass.py", line 81, in getPickedDepth
2021-11-02 08:44:19,193 - CRITICAL - [MainThread] cura.CrashHandler.init [69]: distance = output.pixel(px, py) # distance in micron, from in r, g & b channels
2021-11-02 08:44:19,193 - CRITICAL - [MainThread] cura.CrashHandler.init [69]: TypeError: arguments did not match any overloaded call:
2021-11-02 08:44:19,193 - CRITICAL - [MainThread] cura.CrashHandler.init [69]: pixel(self, QPoint): argument 1 has unexpected type 'float'
2021-11-02 08:44:19,193 - CRITICAL - [MainThread] cura.CrashHandler.init [69]: pixel(self, int, int): argument 1 has unexpected type 'float'`

### Expected results

Support blocker cube is inserted into the view

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

PyQt version: 5.15.

Attached patch seems to fix the problem.
[fix_support_blocker_crash.patch.txt](https://github.com/Ultimaker/Cura/files/7509880/fix_support_blocker_crash.patch.txt)
5

karl-nilsson, 2021-11-10:
Also, there's a 4.12 branch in Ultmaker/fdm_materials, but no corresponding 4.12 tag

fvrmr, 2021-11-12:
Hi @karl-nilsson thank you for your report. Does this also happen in the AppImage of Cura? 

About the tag I will check with the team! Thanks :)

karl-nilsson, 2021-11-12:
@fvrmr no, the AppImage version seems to work fine.
Thanks!

Ghostkeeper, 2021-11-17:
That seems to be a packaging issue from the maintainer then. If the AppImage doesn't crash but Void Linux' package does crash, that means that Void Linux must be using different code or different (versions of) dependencies. Maybe Uranium is outdated or too modern?

We currently only support the AppImage. If there are issues with other versions of Cura's dependencies we'll gladly merge fixes for them, but we don't have the manpower to support anything else than the AppImage for the Linux community by ourselves. So I'll have to close this issue.

closed 2021-11-17 by Ghostkeeper

karl-nilsson, 2021-11-17:
@Ghostkeeper I am the package maintainer. All the dependencies are up-to-date, Uranium is at 4.12.0

The cause of the crash is pretty simple: passing floats to a function that expects integers. Perhaps some layer of the code was performing implicit conversions. If so, that's no longer the case.

Here's the line of code that's failing, as evidenced by the log messages:
https://github.com/Ultimaker/Cura/blob/4.12/cura/PickingPass.py#L81

Here's the patch that fixes the bug:
```diff
diff --git a/cura/PickingPass.py b/cura/PickingPass.py
index 54e886fe6..4d6ef671d 100644
--- a/cura/PickingPass.py
+++ b/cura/PickingPass.py
@@ -72,8 +72,8 @@ class PickingPass(RenderPass):
 
         window_size = self._renderer.getWindowSize()
 
-        px = (0.5 + x / 2.0) * window_size[0]
-        py = (0.5 + y / 2.0) * window_size[1]
+        px = int((0.5 + x / 2.0) * window_size[0])
+        py = int((0.5 + y / 2.0) * window_size[1])
 
         if px < 0 or px > (output.width() - 1) or py < 0 or py > (output.height() - 1):
             return -1
```

Let me know if there's anything else you need from me.

Ghostkeeper, 2021-11-18:
Ah, then maybe the difference is in the PyQt and Qt versions used. Note that Cura doesn't work with the most recent versions of all dependencies. For instance, the most recent version of Qt is 6.2 but it dropped support for most of the interface elements we're creating. And we've had similar issues in the past with Qt 5.13 vs. 5.14, so it's not always just the major version number.

I can see how it could go wrong though if it only accept integers in `QImage.pixel`. That change can be applied to hopefully make it easier for you and other distributions, and for ourselves in the future. It still works correctly in our current version.

hroncok, 2022-01-02:
> Ah, then maybe the difference is in the PyQt and Qt versions used. 

This is most likely a Python 3.10 thing. See e.g. https://github.com/Ultimaker/Uranium/pull/750

hroncok, 2022-01-02:
For more context, Python 3.10 landed some changes that made PyQt and xwPython and other similar wrappers around C(++) GUI libraries no longer accept floats where integers are needed. It is most unfortunate because it requires a lot of rounding/casting everywhere, e.g. https://github.com/kliment/Printrun/pull/1224

I believe it is:

> Builtin and extension functions that take integer arguments no longer accept `Decimals`, `Fractions` and other objects that can be converted to integers only with a loss (e.g. that have the `__int__()` method but do not have the `__index__()` method). (Contributed by Serhiy Storchaka in [bpo-37999](https://bugs.python.org/issue37999).)

From https://docs.python.org/3.10/whatsnew/3.10.html#other-language-changes

karl-nilsson, 2022-01-02:
@hroncok thanks! I was running into the same problem on pythonocc-core, and this helps significantly.

Ghostkeeper, 2022-01-12:
Python 3.10 could well be the culprit, indeed. We are planning to upgrade to Python 3.10 and for that I had to change another of these: https://github.com/Ultimaker/Cura/commit/652043e1d8599460ecf2612042587c46217ef2eb

commit e0e22740 (2021-11-18): Fix picking pixel on integer indices

The QImage.pixel() function should only get integers.
In theory an input position of -2,-2 would get rounded to 0 now. However that shouldn't occur because the user can't click outside of the window. And if it does occur, it's not really a problem either that it's theoretically picking a position inside of the window when you click slightly next to it.

Fixes #10785.