Ultimaker/Cura #10789: I can't authenticate

Evildick, 2021-11-10:
### Application Version

4.11 and 4.12

### Platform

Mac - Catalina

### Printer

Ender 3 V2

### Reproduction steps

After the latest updates, the application no longer allows to authenticate to the site. It closes constantly, making it possible to use it without being authenticated.

### Actual results

After the latest updates, the application no longer allows to authenticate to the site. It closes constantly, making it possible to use it without being authenticated.

### Expected results

After the latest updates, the application no longer allows to authenticate to the site. It closes constantly, making it possible to use it without being authenticated.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

After the latest updates, the application no longer allows to authenticate to the site. It closes constantly, making it possible to use it without being authenticated.

fvrmr, 2021-11-11:
Hi @Evildick thank you for your report. 
Could you please share your log file with me? You can find your log here: `$HOME/Library/Application Support/cura/<Cura version>/cura.log`

Z122-JPL, 2021-11-11:
Hi, @fvrmr  & @Evildick , I have the same problem till Cura 4.9.
The latest version which works without this 'Sign In' problem is Cura 4.8.0 

- **Application version** : Cura 4.9.0 , Cura 4.10.0, Cura 4.11.0

- **Platform** : iMac 21.5' mid-2010, under OS10.13.6  

> @fvrmr , FYI , I can't find any '_cura_' folder under '_$HOME/Library/Application Support/_'
> the log file you requested is probably somewhere under '_Macintosh HD/Applications/Ultimaker Cura/Content/_ " but I can't find it.
> Nevertheless here are following attached here copy/paste in TXT files of the 'Error traceback' and 'Logs' mentioned on three same popup superposed windows when the crash appears.
                               
[Error traceback.txt](https://github.com/Ultimaker/Cura/files/7520380/Error.traceback.txt)
[Logs.txt](https://github.com/Ultimaker/Cura/files/7520381/Logs.txt)

I hope this feedback will help the Cura Team to fix this very annoying problem.

Z122-JPL, 2021-11-13:
@fvrmr & @Evildick , 
The following solution was sent to me by @RoelVossen from Ultimaker who helped me throught a ticket request to fully fix the "Sign In" issue. It seems that this is a very well know issue for all Mac OS. 

> "there is a known issue on OSX where the access control is sometimes not properly set in the **Keychain** application. Could you check whether the **access control** for both 'access_token' and 'refresh_token' of Cura to is set to '**allow all application to access**' this item like below? If this is not the case then please change them accordingly."
[image]

He also told me how to find the request log file : 
https://support.ultimaker.com/hc/en-us/articles/360016538000-Saving-Ultimaker-Cura-logs

fvrmr, 2021-11-16:
Thank you @Z122-JPL 

@Evildick could you check if this works for you!

Evildick, 2021-11-16:
Hi,

Sorry for my delay.
[image: Captura de Tela 2021-11-16 às 19.26.51.png]

The problem remains, I'll try to detail here what happens.

When I try to login with my account, it redirects to the website, where I
can authenticate without problems.

[image: Captura de Tela 2021-11-16 às 19.30.13.png]

Then the application closes and this error message appears.

[image: Captura de Tela 2021-11-16 às 19.28.01.png]

Copy the entire error from the screen above to the attached document.

Ghostkeeper, 2021-11-17:
@Evildick Github allows reply via e-mail but won't include the attachments from those e-mails. We can't see the images you tried to send nor the attached document. Please post them on the Github website itself: https://github.com/Ultimaker/Cura/issues/10789

Ghostkeeper, 2021-11-17:
From the error log provided by Z122-JPL it appears like this is a duplicate of https://github.com/Ultimaker/Cura/issues/9765. But maybe Evildick is experiencing something else. Log files would clarify, if provided.

closed 2021-12-01 by no-response[bot]