Ultimaker/Cura #10790: 4.12 - Crash On Preview - Switching from Skirt (printed/exported already) to brim then slicing then preview caused crash

tyeth, 2021-11-10:
### Application Version

4.12.0 - Released 09-11-21

### Platform

Windows 11

### Printer

Snapmaker 2.0 A350

### Reproduction steps

Import faceguard stl, setup profile with various options like bridging, no infil, supports (tree) etc.
Slice and export Gcode, printed using luban to transfer gcode to printer then printed on touchscreen.
Change to Brim
Slice
Preview
CRASH before/as (~100%) preview fully rendered

### Actual results

CRASH before/as (~100%) preview fully rendered

### Expected results

Preview shown (so I can verify brim not INSIDE object)

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[logs_n_cfg.zip](https://github.com/Ultimaker/Cura/files/7515519/logs_n_cfg.zip)

Do not use the project file as it was created after the crash to comply with the sometimes not relevant project file requirement when filing bug reports.

I repeat, do not use the project file. USE THE STL AND RECREATION STEPS WITHOUT A SAVED/LOADED PROJECT
[stl_n_proj.zip](https://github.com/Ultimaker/Cura/files/7515534/stl_n_proj.zip)

tyeth, 2021-11-10:
If this is related to SmartSlicer, apologies, I had removed it on one pc but the syncing went awry. Trying to uninstall now (marketplace dialog is so slow it becomes unresponsive when trying to do things).

fvrmr, 2021-11-11:
Hi @tyeth thank you for your bug report.
I could not reproduce your crash. Could you let me know if this doesn't happen if you don't have SmartSlice anymore?
We are currently redesigning the Marketplace, so hopefully this will be fixed soon.

You can also manually remove plugins. Go to `C:\Users\<your username>\AppData\Roaming\cura\4.12\plugins`

tyeth, 2021-11-13:
Not noticed this since but not really used it, happy to close for now

closed 2021-11-13 by tyeth