Ultimaker/Cura #10791: std::bad_alloc

dingchow, 2021-11-11:
### Application Version

4.11

### Platform

Windows 11

### Printer

Anycubic Chiron

### Reproduction steps

1. open stl
2. slice
3. fail

### Actual results

std::bad_alloc error in log

### Expected results

finish slice

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7516911/cura.log)
[https://www.thingiverse.com/thing:5028701/files](url) **x285** size

fvrmr, 2021-11-11:
Hi @dingchow thank you for your bug report.
Does this happen for every model you want to slice? Could you also share a project file? With your stl I could not reproduce your issue.

Ghostkeeper, 2021-11-18:
Could be that you're running out of memory? This is the error the engine would give when it runs out of RAM.

closed 2021-11-25 by no-response[bot]