Ultimaker/Cura #10792: Single core being halted by the OS

neofuture, 2021-11-11:
### Application Version

4.12

### Platform

macOS M1 Montery beta 12 

### Printer

a few

### Reproduction steps

Try slicing something complicated and large

### Actual results

Slice does not happen as the OS stalls the process that is abusing the single thread, slice does not complete, I watched activity monitor and after around 20 minutes the thread was locked and the core was returned to the OS

### Expected results

slice to complete 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

none

fvrmr, 2021-11-11:
Hi @neofuture thank you for your report.
Could you share a project file (file - save project) and log file?
You can find your log file here: `$HOME/Library/Application Support/cura/<Cura version>/cura.log`

Ghostkeeper, 2021-11-18:
Since AppleClang doesn't support OpenMP, it's going to be single-threaded on MacOS indeed (at least until we've fully transitioned to a different threading library).

I wouldn't expect the operating system to halt a process though. Why would MacOS stop a process after 20 minutes?

closed 2021-11-25 by no-response[bot]