Ultimaker/Cura #10793: Issues printing with Klipper in Cura 4.12

RyushoYosei, 2021-11-11:
### Application Version

4.12

### Platform

Windows 10

### Printer

Creality Cr-6 SE

### Reproduction steps

Sliced a multi part multiple retraction print in Cura, using Lightning infill

### Actual results

Tried to print, print failed int he en, as it was over-doing  retractions to the point it either jams  up the printer, or just retracts too far and stops w orking.

### Expected results

Should of printed fine, just like it had been for every single print in 4.11

Lightning infil I believe causes so -many retractions- for some reason, that it actually might be causing grinding and other issues, as the print time for lightning infill is also -double- or more what it is for just normal lines infill, saying prints take 8 hours rather then the 2-3 they take on any other infill.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[CCR6SE_Ostrich1.zip](https://github.com/Ultimaker/Cura/files/7517128/CCR6SE_Ostrich1.zip)
This is the file, I don't get it, I t might be a bug with lightning infill or something.

RyushoYosei, 2021-11-11:
Also, the file comes from a 3d printing professor "PrintABlok" set, bu t this has always failed with every attempt using the lightning infill, I'm going to see if it will work with the lines infill I was always using previously.

fvrmr, 2021-11-11:
Hi @RyushoYosei thank you for your report. 
Could you share your project file where you've set the lines infill?
And could you also post some pictures of the result?

I do see in preview that the lightning infill is printed multiple times.

RyushoYosei, 2021-11-11:
The project file is uploaded with th e original post, I can see about trying to dig out some of the previous failed prints to take a picture a bit later when I am more awake, as I just roled back to cura 4.11  so I could get a sucessfulll print.

fvrmr, 2021-11-15:
I will bring this up for discussion with the team. Keep you posted!

fvrmr, 2021-11-16:
I have discussed it with the team. And we have a ticket on our backlog related to this. 
Developers see CURA-8633

rburema, 2021-11-16:
I'm looking into this right now, as it is a potentially serious issue in a main feature of 4.12

I think it might've something to do with meshes that have multiple parts in and of themselves (as opposed to being multiplied, or just different meshes placed together on the buildplate).

I also found a workaround: If you download the 'Mesh Tools' plugin from the Marketplace (no login is required), then install (and restart Cura), (load and) select the model & then select Extensions > Mesh Tools > Split Model into Parts.

rburema, 2021-11-16:
I think I may have a fix for it: https://github.com/Ultimaker/CuraEngine/pull/1530

We should decide if this is worthy of a patch or not this week.

rburema, 2021-11-17:
We do consider this a serious issue and we're likely to make a patch version very soon.

Devs: I also made internal ticket CURA-8702, since CURA-8633 didn't cover the full story.

fvrmr, 2021-11-18:
This is fixed in the 4.12.1 release. You can download it here: https://ultimaker.com/software/ultimaker-cura
I will close this issue since it is fixed!

closed 2021-11-18 by fvrmr

RyushoYosei, 2021-11-26:
I need to re-upen this, as it's not completely fixed, though prints do now finish, it is still causing major issues for Klipper firmware.
The issue is prints that would complete with no issue from Cura 4.11, the exact same settings profile in cura 4.12 now suffers issues, 
Ontop of that, to succeed , because of how 4.12 messes with speeds and accelerations in the Gcode itself as pa rt of the anti-ringing feature, it actually will make klipper skip s  teps and fail prints, ..as I had to reduce my "Max acceleration" values by over half, from 5000, with no issues, down to  2000 to actually not skip with any cura 4.12 prints.
That is not an issue in cura 4.11 STL's on the exact same printer with the same settings for klipper, 

Aka same Klipper setings, Success in 4.11, Failure with 4.12, ..or significant is sues including under extrusion, layers not adhering from the under extrusion including being so visible there are small spots that are holes in the outer skin....

The settings that cura has might be good for Marlin firmware printers, but since klipper reads the exact same gcode, I reccomend you all add a "Klipper" Firmware setting, that disables the changes its making to the gcode in 4.12, since those ch anges cause problems with klippers built in "prevention" of ringing and extrusion issues, and even without those, has issues with cura's settings for the acceleration changes.

Ghostkeeper, 2021-12-06:
If Klipper firmware needs different acceleration setting values to print well, you'd need to make a profile to change those acceleration settings. It's not something we can really help with, given that we don't run Klipper firmware nor have access to your printer to test on.