Ultimaker/Cura #10794: Layer View Slider

donnieval4life, 2021-11-11:
### Application Version

4.12.0

### Platform

macOS Monterey (21A559)

### Printer

Ender 3 Pro

### Reproduction steps

1) Open Cura
2) Slice a model
3) Select preview for layer view
4) Adjust layer slider

### Actual results

When adjusting the layer view slider with the settings dropdown open, it either takes several seconds for the slider to respond to curser movement, or there is no response at all. When the menu is collapsed, the slider is more responsive. The same steps were followed in 4.11.0, 4.10.0, and 4.9.0. 4.9.0 was unaffected by this issue, but 4.10.0 and 4.11.0 were affected in addition to 4.12.0. This slow UI bug seems to extend to the settings dropdown menu, as expanding or collapsing menu items is quite slow in 4.10, 4.11.0, 4.12.0. 4.9.0 is unaffected.
This issue is not limited to any specific sliced model. 50 different models were teste
[cura.log](https://github.com/Ultimaker/Cura/files/7517254/cura.log)
d.

### Expected results

The slider should follow the motion of the curser when clicked and dragged. Dropdown menu items should open without excessive lag when clicked (Again, 4.9.0 response times are noticeably more reasonable.)

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

I am also attaching my 4.9.0 log fil
[cura.log](https://github.com/Ultimaker/Cura/files/7517256/cura.log)
e.

fvrmr, 2021-11-12:
Hi @donnieval4life thank you for report. 
I tried myself on Monterey with 4.12 but I had no issues. Did you try this in 4.9 and 4.12 with the same model?
Could you maybe share your project file?

Ghostkeeper, 2021-11-18:
The next version of Cura will include a performance optimisation for layer view that may help. It prevents the need to re-send vertex data to the GPU with every render.

I don't know if there's really a lot we can do about the performance when the settings list is open though. The settings list has a couple hundreds of settings with a few interface elements each. We did improve performance of that in 4.10 a bit, where we prevented a re-draw of the GUI when only the 3D scene changed, but in your case the layer view slider changes so Qt will re-draw the GUI. We have no real control over what parts of the GUI get re-drawn.

I think we'll wait until the release that contains [this rendering performance update](https://github.com/Ultimaker/Cura/pull/10259).

donnieval4life, 2021-11-19:
4.12.1 has eliminated this issue for me.

fvrmr, 2021-11-22:
Hi @donnieval4life thank you letting us know that 4.12.1 fixed your issue.
I will close this issue since it is now fixed!

closed 2021-11-22 by fvrmr

Ghostkeeper, 2021-11-26:
The performance optimisation is only in 4.13 though, not 4.12.1. 4.12.1 just has a bugfix for the arranger and for lightning infill.