Ultimaker/Cura #10796: Minimum Bridge Extension Length

dlinders10, 2021-11-11:
### Is your feature request related to a problem?

I have issues when printing bridges in Cura because the bridge line does not extend past the walls of the print. This causes the a model with fewer walls to have much worse bridges than the same model with more walls set. In Prusa Slicer, the bridge lines extend past the walls into the infill so the print head does not slow down as much when approaching the walls and it also will get up to a higher speed when changing directions going back the other way. I get almost horizontal bridges in Prusa Slicer while Cura leads to a lot of sagging or bridge lines that break off when the nozzle changes direction since the anchoring area is so short. I have attached photos of the difference in the way the two slicers handle bridges. The only way around this is to up the walls to a high number like 8 so that they extend into the model far enough to slow down later and speed up sooner and anchor better within the print. 

### Describe the solution you'd like

I am suggestion a setting to be added that allows a value  to be set for a minimum value the printer has to travel past the first wall so that it does not prematurely slow down or not have enough room to speed up when changing direction. Even though the extra distance may make the bridge extend into unsupported infill, the extra distance for acceleration and deceleration is what is important. I know there are some areas in the model that will not allow for this like a hole right next to a bridge, but in general this new setting would be very useful.

### Describe alternatives you've considered

I have tried changing the current settings but I can never match the results of prusa slicer with the current settings.

### Affected users and/or printers

Any model with bridges.

### Additional information & file uploads

[image]
[image]

fvrmr, 2021-11-15:
Hi @dlinders10 thank you for your feature request.
I will bring this up with the team. Keep you posted!

fvrmr, 2021-11-16:
I have discussed it with the team. And we decided to defer this request. Because this would be a big change for us to implement and this is not a priority for us since the Bridging settings are also in experimental.
I hope you understand!