Ultimaker/Cura #10800: Cura stuck at loading build volume

topper5282, 2021-11-11:
### Application Version

4.12.0

### Platform

Windows 11

### Printer

None

### Reproduction steps

1. Fresh install of Cura 4.12.0 with no other versions installed and all appdata deleted
2. open software using .exe file

### Actual results

Software gets suck loading build volume and hangs there going no further

### Expected results

Software should open as normal

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7519841/cura.log)

topper5282, 2021-11-11:
After Reseting all network adapters and reinstalling my graphics software and drivers it seems to have resolved the issue

closed 2021-11-11 by topper5282

Ghostkeeper, 2021-11-18:
After loading build volume it's expected to create the main window. If the graphics drivers were an issue, maybe it froze there.