Ultimaker/Cura #10801: Multiply Object crash

zlice, 2021-11-11:
### Application Version

4.11, 4.12

### Platform

Void Linux

### Printer

N/A

### Reproduction steps

1. Load stl
2. Select model
3. Ctrl+M (multiply)
4. Hit Enter (default 1 copy)

### Actual results

```
2021-11-11 10:43:34,684 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]: Traceback (most recent call last):
2021-11-11 10:43:34,685 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:   File "/usr/lib/python3.10/site-packages/UM/JobQueue.py", line 133, in run
2021-11-11 10:43:34,685 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:     job.run()
2021-11-11 10:43:34,685 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:   File "/usr/lib/python3.10/site-packages/cura/MultiplyObjectsJob.py", line 79, in run
2021-11-11 10:43:34,685 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:     group_operation, not_fit_count = createGroupOperationForArrange(nodes,
2021-11-11 10:43:34,685 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:   File "/usr/lib/python3.10/site-packages/cura/Arranging/Nest2DArrange.py", line 119, in createGroupOperationForArrange
2021-11-11 10:43:34,685 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:     found_solution_for_all, node_items = findNodePlacement(nodes_to_arrange, build_volume, fixed_nodes, factor)
2021-11-11 10:43:34,685 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:   File "/usr/lib/python3.10/site-packages/cura/Arranging/Nest2DArrange.py", line 94, in findNodePlacement
2021-11-11 10:43:34,685 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:     converted_points.append(Point(point[0] * factor, point[1] * factor))
2021-11-11 10:43:34,685 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]: TypeError: Point(): arguments did not match any overloaded call:
2021-11-11 10:43:34,686 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:   overload 1: argument 1 has unexpected type 'numpy.float64'
2021-11-11 10:43:34,686 - ERROR - [JobQueueWorker [3]] UM.JobQueue.run [135]:   overload 2: argument 1 has unexpected type 'numpy.float64'
```

### Expected results

Copy object

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

Patch

```
diff --git a/cura/Arranging/Nest2DArrange.py b/cura/Arranging/Nest2DArrange.py
index c29a0648d..dad67ba16 100644
--- a/cura/Arranging/Nest2DArrange.py
+++ b/cura/Arranging/Nest2DArrange.py
@@ -91,7 +91,7 @@ def findNodePlacement(nodes_to_arrange: List["SceneNode"], build_volume: "BuildV
 
         if hull_polygon is not None and hull_polygon.getPoints() is not None and len(hull_polygon.getPoints())>
             for point in hull_polygon.getPoints():
-                converted_points.append(Point(point[0] * factor, point[1] * factor))
+                converted_points.append(Point(int(point[0] * factor), int(point[1] * factor)))
             item = Item(converted_points)
             item.markAsFixedInBin(0)
             node_items.append(item)
```

fvrmr, 2021-11-12:
Hi @zlice thank you for your report. 
Does this also happen with the AppImage of Cura?

ct0br0, 2021-11-12:
It was, just pulled the new one and it is not.

zlice, 2021-11-12:
It was, I downloaded the latest just just and it was fixed

Ghostkeeper, 2021-11-18:
The suggested patch was already applied, back in release 4.9: https://github.com/Ultimaker/Cura/commit/39287264d987d1148c40a515c4e2842d71db20ba
I don't know why it was still an issue apparently in 4.11 and 4.12 and the latest (which is still 4.12) fixed it. But I'm glad to hear that it's fixed now. Perhaps something was indeed still using an older version.

Ghostkeeper, 2021-11-18:
4.12 is still the latest version. If it was originally happening with 4.12.0 and you downloaded the latest (4.12.0) then there shouldn't be any difference. So I think the original issue was reported for the Void Linux package, wasn't it? There was a [similar issue](https://github.com/Ultimaker/Cura/issues/10785) there.

I think the issue is that Void Linux is using a newer version of SIP than we are. We're on 4.19.24, but the latest is 4.19.25. In any case, the missing integer cast was indeed wrong, so I fixed it. Hopefully that'll prevent such bugs in the future when we update SIP (or whichever dependency was causing it).

commit d6f8984c (2021-11-18): Use integer positions for fixed items too

Pynest2d requires it. It seems the SIP version we're using is lenient for this, but newer SIP versions are not.

Fixes #10801.
closed 2021-11-18 by Ghostkeeper