Ultimaker/Cura #10804: Multiply selected and Arrange selected are not working

amitshm, 2021-11-11:
### Application Version

4.12

### Platform

macOS 12.0.1

### Printer

Ender 3 v2

### Reproduction steps

just uploaded stl file
tried multiply and arrange option with no outcome.

### Actual results

a popup of placing objects with on progress bar was shown applying the multiply function 

### Expected results

I should get multiplied objects arranged 

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

I tried several deferent stl files and the results were the same

amitshm, 2021-11-11:
I am looking for Cura 4.11 for macOS with no success 👎

fvrmr, 2021-11-12:
Hi @amitshm thank you for your report this is a duplicate of https://github.com/Ultimaker/Cura/issues/10249
I will close this issue since it is a duplicate.

closed 2021-11-12 by fvrmr