Ultimaker/Cura #10805: Cura want to update disabled plugin

pmarkiewicz, 2021-11-11:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

Unrelated

### Reproduction steps

Disable one of plugins and watch request to update.

### Actual results

Should ignore update

### Expected results

Shoudn't ask to update disabled plug in, this is really annoying especially that I can't click update when plugin is disabled

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

fvrmr, 2021-11-12:
Hi @pmarkiewicz thank you for your report.
I will bring this up for discussion with the team. Keep you posted!

fvrmr, 2021-11-16:
I have discussed it with the team and for us is this expected behaviour. If you don't want the plugin it is better to uninstall the plugin instead of disable the plugin. I hope you understand!

fieldOfView, 2021-11-16:
This plugin in particular cannot be uninstalled (from within Cura atleast), because it ships with Cura.