Ultimaker/Cura #10806: Solidwork Integration

MikeFransch, 2021-11-11:
### Application Version

4.12.0

### Platform

Windows 10 Enterprise

### Printer

Ultimaker S3

### Reproduction steps

Tried to open native Solidworks part files (*.sldprt) in Cura

### Actual results

Nothing got message stating file might be corrupt, inaccessable..

### Expected results

File should open. Solidworks Integration is installed on my PC.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

[image]

MartenPothoven, 2021-11-12:
I have the same promblem

fvrmr, 2021-11-12:
Hi @MikeFransch thank you for your report. 
Could you share your log file? You can find your logfile here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`
Does this happen to every file you want to import?

MartenPothoven, 2021-11-13:
I Run 2 versions of Cura 4.11.0 and 4.12.0

4.11.0 SolidWorksPlugIn works fine.

4.12.0 SolidWorksPlugIn importing WallTest.SLDPRT  >> Unable to Open File

It happens to every file

MikeFransch, 2021-11-14:
> This message is from an external sender. Please be cautious, particularly with links and attachments.

Hi Fenne

Attached pls find the cura log file. And yes since I've install 4.12 this happens every time I try to import Solidworks files. We are running Solidworks 2021.

Regards
Mike Fransch | R & D Technician
[IMDEX Email Signature all brands][Description: logo]
IMDEX Technology NZ | 28B Washdyke Flat Road, Timaru,  New Zealand 7910
Tel:     +64 3 683 9030
Cell:    +64 21 449 583
***@***.******@***.***>
Web:  www.imdexlimited.com<http://www.imdexlimited.com/>

fvrmr, 2021-11-15:
Ah I thought you were using the Native CAD plugin. 
But the Solidworks plugin is not maintained by us. Please reach out to the developer of this plugin: https://thopiekar.eu/cura/cad/solidworks/

I will close this issue since this is something we can not fix.

closed 2021-11-15 by fvrmr

zeleps, 2021-11-18:
For those landing here seeking help, I fixed this issue by opening C:\Users\[username]\AppData\Roaming\cura\4.12\plugins\CuraSolidWorksPlugin\CuraSolidWorksPlugin\SolidWorksReader.py and changing line 470 from:

            if options["sw_previous_active_file"] and "GetTitle" in dir(options["sw_previous_active_file"]):

to

            if options["sw_previous_active_file"]:
           
This applies to plugin version 0.6.1, hopefully the developer will fix this in a newer version. Till then, I hope this helps.

MikeFransch, 2021-11-18:
Thanks, this worked perfectly.

MartenPothoven, 2021-11-19:
Ít works perfect for me.

 

Thanx.

petschedesign, 2022-01-04:
> For those landing here seeking help, I fixed this issue by opening C:\Users[username]\AppData\Roaming\cura\4.12\plugins\CuraSolidWorksPlugin\CuraSolidWorksPlugin\SolidWorksReader.py and changing [...]

Thank you for this!