Ultimaker/Cura #10811: Add more info in the stl header

Juan-ma, 2021-11-12:
### Is your feature request related to a problem?

I think it would be very useful having more info in the header of the stl file like:
- Name of the files/pieces to be printed when printing various pieces at once.
- Estimated printing time in 'hh:mm:ss' format, not only in secconds.
- Materials used.
- Selected printer.
- Print settings.
- Other info available at slicing time on cura.
- Etc.
It would be useful too if it would be configurable through preferences.

### Describe the solution you'd like

Just adding this info in the header of the stl file.

### Describe alternatives you've considered

No alternatives.

### Affected users and/or printers

Everybody.

fieldOfView, 2021-11-12:
This is what project files are for, which save as 3mf files and includes all settings, material, quality and printer used.

STL is not a format that supports metadata likes this. It even only supports a single mesh.

Juan-ma, 2021-11-12:
Ok, maybe not all the info like print settings, etc. but at least the name of the files/pieces included and selected printer. Now stl files includes some info in the header using commented lines, but I think adding this two metadada would be very useful.

fieldOfView, 2021-11-12:
Again, this is what project files are for, not STL files.

Ghostkeeper, 2021-11-19:
To build on fieldOfView's point: We don't need this as a feature for the user, because we already have the ability to restore a print 100% in the form of project files. 3MF is an extensible format and we used that extensibility to extend it to include Cura's information. This includes the names of objects, materials used, printer used, print settings and placement/transformation of objects. It doesn't include printing time; that is restored once the user slices the project. So as a feature we're not going to build it, we already have it.
However I'd be willing to propose something if it is to be standards-compliant, even the optional parts of the STL standard. But is there a standard for this?

STL has two formats: Binary and ASCII.

The Binary format has a header, but it MUST be a fixed-width 80 byte header. It's generally ignored and there are no standards as far as I know of what should be in that header. 80 bytes is not really enough to put a file name or anything like that in, especially internationally with Chinese characters in the UTF-8 encoding. The printing time would be feasible except that the printing time is going to be wrong when printing with different settings (and there's definitely not enough room for the settings) and when using a different version of Cura.

The ASCII format has no header whatsoever and [doesn't define any comment format either](http://www.fabbers.com/tech/STL_Format). The only metadata there can be the name of the solid. We currently write "Uranium STLWriter" followed by the time of exporting there. We could change that to be the print job name, although the print job name is also likely to be the filename itself. Adding any additional information is likely to cause other applications to mark the STLs as corrupt.

Since neither format really has standards for the information you'd like to include, I think we have to reject it. It's not to be able to restore a print in Cura, and it's not to communicate this information to any other application.

closed 2021-11-19 by Ghostkeeper

Juan-ma, 2021-11-19:
Ok, you convinced me with the STL standard argument, so nothing to implement in Cura. I accept it.
Nevertheless I still think that it would be useful including the names of the parts at least in the STL header, not in Cura but in the STL standard, but that's another matter.

Thanks.