Ultimaker/Cura #10813: Crashes on selecting a file

nightmage80, 2021-11-12:
### Application Version

4.11.0

### Platform

Windows 10

### Printer

AnyCubic Chiron

### Reproduction steps

Install Meshmixer (https://www.meshmixer.com/) (I think this is the app that is interfering with the file preview)

Open Cura
Click on the file icon from the main window
Make sure Preview is enabled in the File Dialog
Select a STL file

Cura crashes

Note: If Preview is turned off, it works correctly.

### Actual results

Cura Crashes

### Expected results

File Dialog either shows a preview of the STL file or does not, either is fine.

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[cura.log](https://github.com/Ultimaker/Cura/files/7527599/cura.log)

fvrmr, 2021-11-12:
Hi @nightmage80 thank you for your report.
Do you also have Chitubox installed? If so it is a duplicate of: https://github.com/Ultimaker/Cura/issues/9668

nightmage80, 2021-11-12:
Yes, however, after uninstalling Meshmixer the problem went away.

closed 2021-11-12 by nightmage80