Ultimaker/Cura #10815: AppImage fails to run on Wayland

phrxmd, 2021-11-12:
### Application Version

4.12.0

### Platform

Linux

### Printer

Creality Ender 5

### Reproduction steps

1. Launch Cura AppImage, explicitly specifying Wayland:
```
> QT_QPA_PLATFORM=wayland ./Ultimaker_Cura-4.12.0.AppImage
<frozen zipimport>:259: DeprecationWarning: the imp module is deprecated in favour of importlib; see the module's documentation for alternative uses
qt.qpa.plugin: Could not find the Qt platform plugin "wayland" in ""
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: linuxfb, minimal, offscreen, vnc, webgl, xcb.

/tmp/.mount_UltimafhlYrt/usr/bin//cura.sh: line 14: 13230 Aborted                 (core dumped) cura "$@"
```

### Actual results

Cura fails to start with a message that Wayland is an invalid platform plugin.

Cura will run under XWayland, but only if the QT_QPA_PLATFORM variable is not set.

### Expected results

Cura should start using Wayland UI. 

For comparison, the Cura package of my distribution (OpenSUSE RPM, Cura 4.9.0 as of now) works fine with Wayland. 
This seems to be an issue of the AppImage build.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

No logfile is created.

smartavionics, 2021-11-12:
Hi, I don't have a system running wayland but I have added the Qt wayland client to my Cura build and would be most grateful if you could try running that with QT_QPA_PLATFORM=wayland.

https://github.com/smartavionics/Cura/releases/download/20211110/Cura-mb-master-x86_64-20211110-wayland.AppImage

If it doesn't work, please provide some feedback at https://github.com/smartavionics/Cura/issues/138 which is wayland related.

Thanks.

phrxmd, 2021-11-12:
Thanks @smartavionics - it doesn't work yet, I get a dialogue box complaining about missing OpenGL drivers. Will comment in more detail in the other thread.

fvrmr, 2021-11-16:
Hi @phrxmd thank you for your report.
I checked with two developers but both are not running Wayland. So it is a bit hard for us to reproduce this issue. 
I will leave this issue open and unfortunately we have to wait for more users running into this issue with hopefully more information how to reproduce this.

ajshell1, 2022-01-11:
>I will leave this issue open and unfortunately we have to wait for more users running into this issue with hopefully more information how to reproduce this.

I'm also running into this issue. I'm on Arch Linux, using Sway as my Wayland compositor.

I'm willing to do as much testing as I am asked.

Fortunately, I can state that Cura CAN work under Wayland, as the version provided in the Arch Linux repositories does indeed work for me out-of-the-box under Wayland (using Sway as my compositor). Unfortunately, this packaged version contains a bug (https://github.com/Ultimaker/Cura/issues/10785), so I'd be interested in trying to get the AppImage version to work under Wayland.

Ghostkeeper, 2022-01-18:
If [this](https://aur.archlinux.org/packages/cura-git/) is the one you've been using, then maybe @ad1217 would know what we'd need to change to make it work.

rmsc, 2022-10-12:
> If I substitute `QT_QPA_PLATFORM=wayland` with `QT_QPA_PLATFORM=xcb`, it works.

Unfortunately that doesn't work for me:

```
$ QT_QPA_PLATFORM=xcb cura
realpath: '': No such file or directory
PyInstaller/loader/pyimod02_importers.py:499: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
qt.qpa.xcb: could not connect to display
qt.qpa.plugin: Could not load the Qt platform plugin "xcb" in "" even though it was found.
This application failed to start because no Qt platform plugin could be initialized. Reinstalling the application may fix this problem.

Available platform plugins are: eglfs, linuxfb, minimal, minimalegl, offscreen, vkkhrdisplay, vnc, wayland-egl, wayland, xcb.

/tmp/.mount_Ultima77Tk0g/AppRun: line 16: 19540 Aborted                 (core dumped) $scriptdir/Ultimaker-Cura "$@"
```

jellespijker, 2022-10-12:
> With the new 5.0.0 AppImage, it STILL won't run under Wayland, but the error message is different - it fails with "OpenGL 2.0 required":
> 
> ```
> > QT_QPA_PLATFORM=wayland LD_PRELOAD=/usr/lib64/libstdc++.so.6 ./Ultimaker-Cura-5.0.0-linux.AppImage
> PyInstaller/loader/pyimod03_importers.py:495: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
> qt.qpa.wayland: EGL not available
> ```
> 
> If I substitute `QT_QPA_PLATFORM=wayland` with `QT_QPA_PLATFORM=xcb`, it works.

is this an error message related to the system libc++ version.

which should be addressed by running the modern AppImage based on a newer Ubuntu version

rmsc, 2022-10-16:
> This is an error message related to the system libc++ version.
> 
> which should be addressed by running the modern AppImage based on a newer Ubuntu version

Sorry, but how is this related to the libc++ version? 

In any case I'm on an up-to-date Archlinux system, gcc version 12.2.0-1.

jellespijker, 2022-10-16:
@rmsc which of the two AppImages are you using?

rmsc, 2022-10-16:
The one from Archlinux AUR:
https://github.com/Ultimaker/Cura/releases/download/5.1.1/Ultimaker-Cura-5.1.1-linux.AppImage

I've just noticed there's a **modern** image as well. Tried that one but it failed with a different error:

```
$ cura
PyInstaller/loader/pyimod02_importers.py:499: DeprecationWarning: the imp module is deprecated in favour of importlib and slated for removal in Python 3.12; see the module's documentation for alternative uses
Cyclic dependency detected between "file:///tmp/.mount_curarJbCkX/share/cura/resources/qml/Actions.qml" and "file:///tmp/.mount_curarJbCkX/share/cura/resources/qml/Actions.qml"
QQmlEngine::setContextForObject(): Object already has a QQmlContext
[destroyed object]: error 7: importing the supplied dmabufs failed
The Wayland connection broke. Did the Wayland compositor die?
```

rmsc, 2022-10-16:
Looking at sway's output, here's the problem:

```
00:02:14.697 [ERROR] [wlr] [render/vulkan/texture.c:590] Unsupported pixel format 36314752 (RG16)
```

jellespijker, 2022-10-16:
Unfortunately we can't support Cura verwions distributed by system package managers; That being said, it looks like the AppImage in the AUR is the regular AppImage. Wich would explain your original error, since you have a more modern libc++ on your system then the regular AppImage. That is the whole reason we added an AppImage based on a more modern distro with a newer libstdc++.

When you tried to run the modern AppImage it seems to fail somewhere else. Looking at that particular error I assume that you're running Wayland. Unfortunately we don't support wayland yet. What happens if you try running it on X11?

Since wayland is becoming more mature and KDE might make it the default in the near future I will bring up the discussion with the team, but for now I have to tell you that we can't support your particular Wayland setup.

jellespijker, 2022-10-17:
thank you for testing on X11, I will let you know on the outcome of the Wayland discussion

idlesign, 2022-12-27:
Ubuntu 22.04.1 Wayland 5.2.1 app image for Linux-Modern solved the issue for me.

unclamped, 2023-04-28:
Fedora 38, Wayland. The app image for linux-modern works just fine, and so does the RPM and Flatpak versions of Cura. The only issue I'm experiencing however is that if I use Wayland by adding the `QT_QPA_PLATFORM=wayland` env variable alongside the execution command, it becomes REALLY laggy and the menus glitch a bit, f.e. scrolling doesn't work, but for some reason moving my mouse cursor while using the wheel does work. This occurs on both the AppImage and the Flatpak version, although the RPM version doesn't (but that one's still in Cura 4 for whatever reason lol)

unclamped, 2023-10-27:
@MariMakes Doesn't seem like OP is active in GitHub as a whole anymore... I have moved from Fedora to NixOS, and I no longer use Wayland so I can't tell much

J-Wyz, 2023-10-28:
@MariMakes I installed it with the official 5.5.0 release, which resulted in the same issue

InRiPa, 2024-08-10:
Still an Issue with `5.8.0, using Arch Linux and the AppImage from the website.
```bash
╰─ ./UltiMaker-Cura-5.8.0-linux-X64.AppImage 
qt.qpa.wayland: EGL not available
```

@J-Wyz  `QT_QPA_PLATFORM=xcb` makes it work.

loomsen, 2024-08-25:
Same on Fedora, althought I guess the distro doesn't actually matter. Still an issue with 5.8.0, I vote for reopening.

Flatfire, 2024-10-07:
Chiming in to add that this is still a problem in the appimage for 5.8.1

phrxmd, 2024-10-08:
@Flatfire you can fix it with this workaround (from [this comment](https://github.com/Ultimaker/Cura/issues/19386#issuecomment-2248619478):

Download AppImageTool from https://github.com/AppImage/appimagetool/releases, then issue the following:

```
./UltiMaker-Cura-5.8.1-linux-X64.AppImage --appimage-extract 
cd squashfs-root 
cp /usr/lib/libwayland-client.so.0.23.1 ./libwayland-client.so.0
cp /usr/lib/libwayland-egl.so.1.23.1 ./libwayland-egl.so.1 
mv com.ultimaker.cura.desktop com.ultimaker.cura.desktop.bak
sed "s/.png//" com.ultimaker.cura.desktop.bak > com.ultimaker.cura.desktop
cd ..
ARCH=x86_64 ./appimagetool-x86_64.AppImage squashfs-root 
mv UltiMaker_Cura-x86_64.AppImage UltiMaker-Cura-5.8.1-linux-X64-with-wayland.AppImage
```

This modified AppImage starts fine under Wayland for me.

smartavionics, 2024-10-08:
You should just be able to remove those libraries from the appimage and then the system libraries will be found instead. I think it is a mistake for those libraries to be included in the appimage. I have now removed them from my Cura releases along with quite a few other libraries that still get bundled in the UM releases.

phrxmd, 2024-10-08:
In reply to @smartavionics' comment [from here](https://github.com/Ultimaker/Cura/issues/19724#issuecomment-2399351172):

> You should just be able to remove those libraries from the appimage and then the system libraries will be found instead. I think it is a mistake for those libraries to be included in the appimage. I [...]

I agree that this can be made to work by modifying the AppImage. But a user shouldn't have to either add or remove libraries from an AppImage just to make it work. The whole point of having an AppImage is that it should be independent of what libraries the user has on their system. 

If that can't be guaranteed, then maybe AppImage isn't the best solution for distributing the app and one should look for a different packaging format

smartavionics, 2024-10-08:
As I said above, libraries are being included in the UM appimage that really don't need to be there. The UM developers need to weed out all the libs that are already installed on the target systems.

phrxmd, 2024-10-09:
That's correct; since on a Wayland system you can expect the Wayland libs to be present, the following workaround works for me as well:

```
./UltiMaker-Cura-5.8.1-linux-X64.AppImage --appimage-extract 
rm squashfs-root/libwayland* 
sed -i.bak "s/.png//" squashfs-root/com.ultimaker.cura.desktop
ARCH=x86_64 appimagetool squashfs-root 
mv UltiMaker_Cura-x86_64.AppImage UltiMaker-Cura-5.8.1-linux-X64-with-wayland.AppImage
```

Flatfire, 2024-10-09:
> @Flatfire you can fix it with this workaround (from [this comment](https://github.com/Ultimaker/Cura/issues/19386#issuecomment-2248619478)):
> 
> Download AppImageTool from https://github.com/AppImage/appimagetool/releases, then issue the following:
> 
> ```
> ./UltiMaker-Cura-5.8.1-linux-X64.AppImage --appimage-extract 
> cd squashfs-root 
> cp /usr/lib/libwayland-client.so.0.23.1 ./libwayland-client.so.0
> cp /usr/lib/libwayland-egl.so.1.23.1 ./libwayland-egl.so.1 
> sed -i.bak "s/.png//" com.ultimaker.cura.desktop
> cd ..
> ARCH=x86_64 ./appimagetool-x86_64.AppImage squashfs-root 
> mv UltiMaker_Cura-x86_64.AppImage UltiMaker-Cura-5.8.1-linux-X64-with-wayland.AppImage
> ```
> 
> This modified AppImage starts fine under Wayland for me.

At the moment, I'm just doing what I do with most AppImages, and dumping it in a folder with an applications menu shortcut that has any env variables I might need. Using `QT_QPA_PLATFORM=xcb` is currently doing the trick without needing to modify the AppImage itself.