Ultimaker/Cura #10817: Tree support line width

ais80, 2021-11-12:
### Is your feature request related to a problem?

Tree supports printed at higher line widths are very solid and hard to remove

### Describe the solution you'd like

Ability to adjust the line width for supports

### Describe alternatives you've considered

I cant see one

### Affected users and/or printers

All users

Asterchades, 2021-11-13:
There's already an option for Support Line Width. It's with all of the other line width settings up in Quality, but is only visible when Supports are enabled.

You could also adjust the Support Flow in the Material section. Again this is only visible when Supports are enabled.

ais80, 2021-11-13:
Thank you so much. I never noticed that feature before

fvrmr, 2021-11-16:
Hi @ais80 thank you for your request. 
I will close this issue since there is a solution for this!

closed 2021-11-16 by fvrmr