Ultimaker/Cura #10818: Drop-down menu "Add a networked printer" is not available.

joe-ercolino, 2021-11-12:
### Application Version

4.12.0

### Platform

Mac OS 12.0.1

### Printer

Ohter

### Reproduction steps

1. Install Ultimaker Cura 4.12.0
2. ...
3. Add a printer
4. Click on "Add a networked printer" menu

### Actual results

The menu "Add a NON-networked printer" menu is displayed.

### Expected results

The menu "Add a networked printer" menu should be displayed.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Can not continue with because the actual printer is not listed and can not be selected.

[image]

[image]

fieldOfView, 2021-11-12:
In your first image, the "Add a networked printer" section is already showing. 

However the "Add a networked printer" is mostly just meant for Ultimaker 3, S5 and S3 printers at this point. So that is probably why your printer is not listed.

fvrmr, 2021-11-15:
Hi @joe-ercolino thank you for your report.
What printer do you want to add? Like @fieldOfView mentioned this is for Ultimaker printers.

closed 2021-11-29 by no-response[bot]