Ultimaker/Cura #10819: BUG: multiple instances of cura can startup instead of opening in the same window

darrenoakey, 2021-11-12:
### Application Version

4.12.0

### Platform

Windows 10

### Printer

wanhao mini

### Reproduction steps

* double click on an stl - get an instance of cura
* print
* double click on an stl - get ANOTHER instance of cura
* the second instance can't print via usb, because the first instance is locking the serial port
* close the first instance
* the second instance still can't print via usb

### Actual results

* you got multiple instances of cura.
* you can't print from the second instance
* you have to restart all instances manually to then be able to print again

### Expected results

You don't get multiple instances.    

I'd argue that unless you have multiple printers, (and probably 99% of users don't) there's no use case for multiple instances - you could argue you want to setup the next print - but that's just not valid - because 
- it doesn't take long to put in parms
- there's a large overhead in changing over to the next print which dwarfs the time in the app
- you can't actually print from the second instance

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

speaking as a programmer, this is pretty trivial to implement, - on startup you check if cura is running, and if it is, send a message to it to open the file - and there's a million ways to do _that_.   
Even if you don't agree that less than 1% of people would ever want the current behaviour - please at least add a setting to go into single instance mode.

fieldOfView, 2021-11-12:
There's a preference for that.
On the "General" pane of the Preferences, look in the "Opening and saving files" section and you will find a setting labeled "Use a single instance of Cura". From Cura 4.12 onwards, there's also a toggle to "Clear buildplate before loading model into the single instance" that you may find useful.

> speaking as a programmer, this is pretty trivial to implement, 

As a programmer you should know that implementing something in one framework may be more trivial than implementing it in another; It can be fairly annoying if people tell you something is pretty trivial to implement. Not in this case though; now I can say "speaking as someone with working eyesight, it is pretty trivial to find this preference".

darrenoakey, 2021-11-13:
Thank you.  I even appreciate the snark :)

fieldOfView, 2021-11-13:
At the risk of over-snarking, you may close the issue ;-)

(I'm a 3rd party, who does not have the privileges to close issues)

closed 2021-11-13 by darrenoakey