Ultimaker/Cura #10824: [4.12] Monotonic should finish a skin when possible before jumping to another.

printingotb, 2021-11-13:
### Application Version

4.12

### Platform

win10

### Printer

Custom fff

### Reproduction steps

See attached project and pictures. Monotonic is enabled for top/bottom, top surface and ironing.

### Actual results

Top/bottom layers and top surface skin layers are printed as expected, one area is finished at a time.
However when it starts ironing, area A isn't finished and it jumps to area B halfway through. 
This produces unnecessary scars.

### Expected results

I'd expect it to finish any area that can be printed in one continuous pass. Naturally, the area with the 20 can't but area A should.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[image]
(print in photo below is printed in a different orientation but same result).
[image]
[monotonic.3mf.zip](https://github.com/Ultimaker/Cura/files/7531744/monotonic.3mf.zip)

printingotb, 2021-11-13:
Duplicating the model changes the fill behavior. The left part prints as expected, finishing all areas completely in one pass. The right one finishes doesn't finish the lower right wedge.
Changing print order to one at a time changes the behavior again.
[image]

jellespijker, 2021-11-15:
@fvrmr this might be related to CURA-8692

fvrmr, 2021-11-15:
Hi @printingotb thank you for your bug report. 
I will bring this up with the team. Keep you posted!

fvrmr, 2021-11-16:
I have discussed it with the team and it is not related to the ticket @jellespijker mentioned but to CURA-8474
I will link this issue to that ticket!

SniderThanYou, 2022-01-27:
I'm encountering the same thing:

[image]

Gerhard-59, 2024-02-11:
Hello, Is there any pogress on this issue, as it is still present with ironing.
See enclosed screenshot and poject.

[image]
[PET_Coin-for-4mm-PTFE_0h19m.zip](https://github.com/Ultimaker/Cura/files/14231387/PET_Coin-for-4mm-PTFE_0h19m.zip)

GregValiant, 2024-11-16:
Is this still a problem in current versions of Cura (5.8.0 and up)?  Can this be closed?

Gerhard-59, 2024-11-16:
> Is this still a problem in current versions of Cura (5.8.0 and up)? Can this be closed?

Actually I don't use cura anymore. Please ask someone else.