Ultimaker/Cura #10825: Fuzzy Skin Blocker and More Settings

Kayakbabe, 2021-11-13:
### Is your feature request related to a problem?

 if I'm printing something that has a threaded part that is how it attaches to another object. I want the fuzzy skin everywhere except where that threaded extension is. Or an object that has a screw hole for affixing to something. I want fuzzy everywhere but inside that hole (so the dimensions of that hole are accurate). 
I know there is a setting to just have fuzzy on the outside only or everywhere. But it isn't enough.

### Describe the solution you'd like

What I want is more control AND blockers for fuzzy. 
With fuzzy blockers only (more than one per model), it would be possible, with work, to accomplish a lot.
With a couple more settings AND fuzzy blockers it would be super fast to accomplish a lot quickly.

examples that could be done:

1. fuzzy on outside and top skin
2. fuzzy everywhere except holes under a certain diameter
3. combining 1 and 2 with a fuzzy blocker means control over exceptional cases which are not covered by the above 2 settings. Example an open channel which holds another part will fit and not be affected by the dimensional inaccuracy that fuzzy skin induces.

(see attached images)

### Describe alternatives you've considered

I don't know of any alternatives that don't require a LOT of post-processing after 3D printing. 

### Affected users and/or printers

Anyone making objects that need to be visually pleasing yet still functional. Anyone making prototypes, one-offs,  and molds for mass production who want a skin on the 3d printing item that doesn't require post processing. 

This would be great for parts which are visible to the public, I'm thinking boxes which hold electronics like raspberry pi and small microcomputers affixed to displays, gift objects, trophies, the songbird record player I'm making presently, puzzles, other electronics, cases, frames, etc. so many things!

### Additional information & file uploads

The zone in orange is where I need extra control. 
It has screw holes and also an open channel that needs to not be fuzzy. But the rest of the part would have a really awesome appearance if it was fuzzy everywhere else. The rest of the project it goes with would greatly benefit from the same treatments.

[image]
[image]

I snippedimages  from Frame Theory 3D SongBird Mk.3+, myminifactory , I think it's okay to put these images on this forum and it's okay with them if I attribute it. It's not my design.

fvrmr, 2021-11-15:
Hi @Kayakbabe thank you for your detailed feature request!
Fuzzy skin will be very complex to implement. See also https://github.com/Ultimaker/Cura/issues/9246
About the fuzzy skin blocker, I will discuss this with the team. Keep you posted!

fvrmr, 2021-11-16:
I have discussed it with the team and the best work around to do this is with a cutting mesh. 
For example: 
[image]

Here you can read more about cutting meshes: https://support.ultimaker.com/hc/en-us/articles/360013370140-How-to-adjust-print-settings-of-a-part-of-my-model-in-Ultimaker-Cura

I will close this issue since this is something we are not going to implement. But let me know if you have more questions or ideas!

closed 2021-11-16 by fvrmr

C0unt23r0, 2024-03-17:
Hi, I know this is an old request/thread on this topic. I’ve tried the mentioned solution but couldn’t achieve what I was attempting.

The problem: Imagine an object with a hole and fuzzy skin activated. I don’t want to have fuzzy skin inside the hole. You might suggest, “Why not activate ‘Outside Only’?” However, I haven’t placed the object flat on the surface. It’s positioned at a certain angle and has built-in supports on the print bed. While this allows the top surface to receive fuzzy skin, it also renders the “Outside Only” option ineffective for some parts.

Returning to the problem, when applying “cutting meshes,” it creates extra walls for the support blocker and the original object with fuzzy skin applied. Unfortunately, this worsens the situation.

Perhaps, three years later, someone might consider that this feature request could now be useful?