Ultimaker/Cura #10826: Build plate temperature initial layer doesn't reflect high temp for Snapmaker 2.0 A350

Kilken005, 2021-11-13:
### Application Version

4.12.0

### Platform

Microsoft Windows [Version 10.0.19043.1288]

### Printer

Snapmaker 2.0 A350

### Reproduction steps

1. Set build plate temperature initial layer higher than build plate temperature
2. Slice
[image]

### Actual results

;Generated with Cura_SteamEngine 4.12.0
M82 ;absolute extrusion mode
M104 S200 ;Set Hotend Temperature
M140 **S50** ;Set Bed Temperature
G28 ;home
G90 ;absolute positioning
G1 X-10 Y-10 F3000 ;Move to corner 
G1 Z0 F1800 ;Go to zero offset
M109 S200 ;Wait for Hotend Temperature ;Wait for Hotend Temperature
M190 **S50** ;Wait for Bed Temperature ;Wait for Bed Temperature
...
;LAYER:1
M140 S50

### Expected results

;Generated with Cura_SteamEngine 4.12.0
M82 ;absolute extrusion mode
M104 S200 ;Set Hotend Temperature
M140 **S70** ;Set Bed Temperature
G28 ;home
G90 ;absolute positioning
G1 X-10 Y-10 F3000 ;Move to corner 
G1 Z0 F1800 ;Go to zero offset
M109 S200 ;Wait for Hotend Temperature ;Wait for Hotend Temperature
M190 **S70** ;Wait for Bed Temperature ;Wait for Bed Temperature
...
;LAYER:1
M140 S50

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

Same error on Cura 4.11.0 but does not reproduce o n other printer e.g. Anycubic 4Max

joaquinabian, 2021-11-14:
I confirm the problem for both _Build Plate_ and _Printing Temperatures_. 
GCode generated doesn't take into account values for _Initial Layer_.
Here on an Snapmaker 250, Normal profile. Cura 4.12.0

GregValiant, 2021-11-16:
In the Snapmaker2.def.json file (which appears to have been written by someone at Snapmaker) the startup gcode is defined that way and neither the A250 nor the A350 specific definitions override it.
M104 S{material_print_temperature} ;Set Hotend Temperature
M140 S{material_bed_temperature} ;Set Bed Temperature
and later:
M109 S{material_print_temperature} ;Wait for Hotend Temperature
M190 S{material_bed_temperature} ;Wait for Bed Temperature

Those keywords need to be "material_print_temperature_layer_0" and "material_bed_temperature_layer_0".  You can alter them in your StartUp G-Code.  You should probably let Snapmaker know as well so they can alter their definition file.

Kilken005, 2021-11-16:
> In the Snapmaker2.def.json file (which appears to have been written by someone at Snapmaker) the startup gcode is defined that way and neither the A250 nor the A350 specific definitions override it. [...]

Thanks @GregValiant that has fixed the issue and I'll log a support request with Snapmaker.

closed 2021-11-16 by Kilken005

tyeth, 2021-11-20:
Please unclose/reopen this issue and we can get more traction on the actual fix in cura rather than hand editting settings files...
#10849

GregValiant, 2021-11-20:
@tyeth, the problem isn't in Cura.  The problem is in the definition file that Snapmaker (or a Snapmaker user in the "community") provided to Ultimaker.

Ghostkeeper, 2021-11-23:
The issue should be fixed by #10849 now.