Ultimaker/Cura #10827: Slicing failed on Mac M1

jorgecorn, 2021-11-14:
### Application Version

4.12

### Platform

Mac OS

### Printer

Elegoo Neptune 2

### Reproduction steps

Try to slice file with any print settings.

### Actual results

Fails

### Expected results

Works

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

https://www.icloud.com/iclouddrive/0rqYI9dNgTex4DuCvXEO1s6Qg#4x_M4_Nut_and_Bolt

fvrmr, 2021-11-15:
Hi @jorgecorn thank you for your bug report.
I could not reproduce your issue. But does this happen with every file?
Could you also share your log file? Here you can find your log file: `$HOME/Library/Application Support/cura/<Cura version>/cura.log`

Ghostkeeper, 2021-11-23:
The project file contains a mod for the printer definition `elegoo_neptune_2`. We'd need that definition to test it properly. How can we install this mod?

closed 2021-11-29 by no-response[bot]