Ultimaker/Cura #10829: 4.12.0 Won't Launch on Windows 10

regarsafari, 2021-11-14:
### Application Version

4.12.0

### Platform

Windows 10

### Printer

N/A

### Reproduction steps

1. Downloaded Cura 4.12.0 from webpage
2. Installed Cura 4.12.0
3. Clicked on shortcut

### Actual results

Nothing happens. The loading circle comes up next to my mouse pointer and then disappears and nothing else happens.

### Expected results

Cura should launch.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

The only version of Cura that seems to work on my computer is 4.8. Every version above that produces the same result. I have tried re-installing Cura multiple times, deleting files, re-installing Visual C++ 2015 Redistributable, and still nothing has worked. I don't seem why only version 4.8 wants to work on my computer. No log files are produced either.

pumberto, 2021-11-15:
A ultima versão que funcionou no meu 2 computadores a versão 4.8. Comigo acontece o seguinte, o programa carrega, consigo instalar as impressoras, fazer configurações, mais u abrir qualquer arquivo STL, o CURA fecha.

fvrmr, 2021-11-15:
Hi @regarsafari thank you for your report.
Could you share your log file? You can find your log file here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

fvrmr, 2021-11-15:
> A ultima versão que funcionou no meu 2 computadores a versão 4.8. Comigo acontece o seguinte, o programa carrega, consigo instalar as impressoras, fazer configurações, mais u abrir qualquer arquivo STL, o CURA fecha.

According to Google translate your Cura closes when opening a STL file, am I correct?
If so this is not related to this issue. Do you have Chitubox installed? If so see: https://github.com/Ultimaker/Cura/issues/9668

pumberto, 2021-11-15:
[cura.log](https://github.com/Ultimaker/Cura/files/7540874/cura.log)
Segue arquiv de LOG

regarsafari, 2021-11-19:
> Hi @regarsafari thank you for your report.
> Could you share your log file? You can find your log file here: `C:\Users\<your username>\AppData\Roaming\cura\<Cura version>\cura.log`

Hello, there is actually no cura.log in that directory.

fvrmr, 2021-11-23:
Hmm, do you maybe have multiple screens?
What does show that directory?

pumberto, 2021-11-23:
Problema resolvido, com a remoção da DLL do chitubox para abrir aas miniaturas.

closed 2021-12-07 by no-response[bot]