Ultimaker/Cura #10834: Top/bottom skin layers between outermost wall layer

Subwing, 2021-11-15:
### Is your feature request related to a problem?

When using a multiple wall count the top skin layer is placed between the innermost walls making all of the walls visible in the top layer. This doesn't look good, especially for thin parts. In addition, I don't like to use "fill gaps between walls" as for thin tall parts it makes the print time considerably longer. However, not using this feature will make these small gaps visible as the top skin layer doesn't cover these.

### Describe the solution you'd like

Make it possible to place the top/bottom skin layers between the outermost walls. This will look much better and in also cover the small gaps between walls that has not been filled when not using "fill gaps between walls".

### Describe alternatives you've considered

An alternative solution would be to make it possible to enable "fill gaps between walls" only for the skin layers. This will fill out small gaps in the top/bottom skin layer, but all of the wall layers will still be visible in the top/bottom layers.

### Affected users and/or printers

Everyone will benefit as the top/bottom layers will have a better look. In addition, those who prints tall, thin parts will benefit greatly with a shorter print time as they don't have to use the "fill gaps between walls" in order to fill out gaps in the top/bottom layers.

### Additional information & file uploads

See attached photo of comparison between a part printed with Cura and another printed with Makerbot Print. Both parts have multiple walls, but the Makerbot Print place the top layer between the outermost walls making for a much better looking top layer. In addition small gaps are completely covered.
[image]
.

fvrmr, 2021-11-16:
Hi @Subwing thank you for your feature request. 
Since Cura 4.11 we have a new setting called Monotonic Ordering. This will give you also a better top skin layer. Did you check that already?
Also our Arachne release uses variable line width to prevent these gaps. You can read here more about Arachne: https://github.com/Ultimaker/Cura/releases/tag/Arachne_engine_beta

Let me know if that could work for you!

Asterchades, 2021-11-17:
I feel like they not only want the higher quality on the top layers, but also to have it that only a single wall is is exposed on the top/bottom rather than having however many walls you've actually designated for the remainder of the print. Sort-of like applying a negative value to the Extra Skin Wall Count, but only for the top-most and bottom-most layers of any given section.

By extension, this would allow for a more consistent top/bottom layer on some objects with narrow portions - but only those still wide enough to allow at least two walls on each side. Printing such areas with a single wall and skin would leave fewer gaps than attempting to print multiple walls and then skin (or gap fill) what remains.

Ghostkeeper, 2021-11-23:
Monotonic ordering of skin doesn't apply here because there is no skin.

I think this is effectively a duplicate of https://github.com/Ultimaker/Cura/issues/7324, although the scope is slightly different. We'll close this one to consolidate the discussion in one thread instead of two.

closed 2021-11-23 by Ghostkeeper