Ultimaker/Cura #10835: Cura Monotonic-Related Settings Not Visible Under Print Settings

MinProwler, 2021-11-15:
### Application Version

4.12 and 4.12beta

### Platform

Macintosh (10.13.6)

### Printer

Custom Delta Printer

### Reproduction steps

1. Enable visibility for all settings
2. Turn on visibility for all settings with "mono" in their name
3. Close settings
4. Quit Cura
4. Restart Cura
5. Search for all settings with "mono" under Print Settings window on right hand side

### Actual results

No settings with "mono" appear under print settings.  See screen shot attached.

### Expected results

Monotonic print settings are not visible in 4.12 and 4.21beta (discovered retrospectively--sorry).  Monotonic print settings are visible in 4.11
[image]
.

### Checklist of files to include

- [ ] Log file
- [ ] Project file

### Additional information & file uploads

In case the screen shot does not appear above, I am repeating here.  Sorry, this my first time tring to do a bug report.
[image]

fvrmr, 2021-11-16:
Hi @MinProwler thank you for your bug report.
Did you use the same profile in 4.11? Could you share your profile or a project file (file - save project) with me?
For me the monotonic top/bottom order is for me visible.

MinProwler, 2021-11-16:
I did use the same profile in 4.11.  I've attached 3mf files saved from Cura 4.12 and 4.11.
Thank you!

[CuraProfiles.zip](https://github.com/Ultimaker/Cura/files/7546421/CuraProfiles.zip)
 I

MinProwler, 2021-11-22:
Fixed!  I switched to my vase profile and noticed the Discard or Keep pop up window included monotonic settings.  I discarded changes and when I switched back to my standard profile (attached above), the monotonic settings were visible.

Unlike CHEP, all my printer profiles transferred ok, fwiw.

Thanks for your help and making such a great product!

fvrmr, 2021-11-22:
Hi @MinProwler thank you for your update! I will close this issue since it is fixed :)

closed 2021-11-22 by fvrmr

fieldOfView, 2021-11-23:
Tip for the next time something like this happens: In the Setting Visibility dialog, the little (!) icon has an explanation why the setting is not currently showing in the settings bar.

fieldOfView, 2021-11-23:
For example, the tooltip of the (!) icon for the `Monotonic Ironing Order` would have shown that the setting is hidden because of the value of `Enable Ironing`; unless you enable ironing, there is no use in changing whether ironing should use the monotonic pattern.