Ultimaker/Cura #10837: Simulation view crash in 4.11 4.12 (and problems in 4.10)

bigRedBlue, 2021-11-15:
### Application Version

4.11 4.12 (and problems in 4.10)

### Platform

win7 Radeon HD 6700

### Printer

Custom (although tested with various Ultimaker profiles as well)

### Reproduction steps

1. Switch to preview mode (made no difference if a model was on the plate, sliced, or plate was empty)
2. Crash occurs

### Actual results

Crash and bug report is generated

```
Traceback (most recent call last):
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Qt\Bindings\MainWindow.py", line 255, in _render
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtRenderer.py", line 163, in render
  File "C:\Program Files\Ultimaker Cura 4.11.0\plugins\SimulationView\SimulationPass.py", line 65, in render
    self._layer_shader.setUniformValue("u_active_extruder", float(max(0, self._extruder_manager.activeExtruderIndex)))
AttributeError: 'NoneType' object has no attribute 'setUniformValue'
Thread 0x000012d0 (most recent call first):
  File "threading.py", line 306 in wait
  File "zeroconf\__init__.py", line 2534 in wait
  File "zeroconf\__init__.py", line 1715 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00001180 (most recent call first):
  File "threading.py", line 306 in wait
  File "threading.py", line 558 in wait
  File "C:\Program Files\Ultimaker Cura 4.11.0\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000012bc (most recent call first):
  File "zeroconf\__init__.py", line 1369 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00000ad8 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.11.0\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00001214 (most recent call first):
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Backend\Backend.py", line 165 in _storeStderrToLogThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00000e54 (most recent call first):
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Backend\Backend.py", line 153 in _storeOutputToLogThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000011a8 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00000cc0 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00000a00 (most recent call first):
  File "threading.py", line 302 in wait
  File "threading.py", line 433 in acquire
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 98 in _nextJob
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\JobQueue.py", line 124 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Current thread 0x00001104 (most recent call first):
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 396 in _logInfoWidget
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 182 in _createDialog
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 86 in __init__
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\bin\cura_app.py", line 159 in exceptHook
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 452 in _show
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 479 in event
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1090 in event
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 452 in _show
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 479 in event
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1090 in event
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 452 in _show
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 479 in event
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1090 in event
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 393 in exec_
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 867 in run
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.11-exe\build\inst\bin\cura_app.py", line 233 in <module>
  File "Console.py", line 36 in run
  File "__startup__.py", line 74 in run
```

### Expected results

Simulation View plugin should have been initialized

### Checklist of files to include

- [X] Log file
- [ ] Project file

### Additional information & file uploads

[276570081_cura4_11crash.log](https://github.com/Ultimaker/Cura/files/7541003/276570081_cura4_11crash.log)
[DxDiag.txt.log](https://github.com/Ultimaker/Cura/files/7541004/DxDiag.txt.log)

This log is from 4.11 because I have 4.12 up and running as I mention below. I can break 4.12 and generate a log if that would help.

Bug initially reported on forums here:
https://community.ultimaker.com/topic/38604-simulation-view-crash-in-411-412-and-problems-in-410/?_fromLogin=1

> Simulation view is crashing Cura in 4.11 and 4.12. It does not crash 4.10, but the object remains shadowed transparent like it had not been sliced.
> 
> I copied the simulation view plugin from 4.9 over the 4.12 plugin and that lets me at least see a preview. XRAY mode still works when I disable the plugin from the marketplace.
> 
> Crashing occurs whether a STL is on the build plate or not, and whether it is sliced or not. I simply can't open the mode.
> 
> Because of the mention of the extruder settings in the bug report, I suspected it was an issue with my custom printer so I did a clean re-install of 4.12 and selected a default Ultimaker printer. This did not fix the problem.
>  
> I am now thinking something in 4.10+ is just not playing nice with my graphics card. This machine is using an AMD Radeon HD 6700 (which runs solidworks great).

fieldOfView, 2021-11-17:
Your GPU driver can not compile the "vertex shader" used to render the simulation view. It is a bit sparse in details, but you could try the following:

See line 112 of `C:\Program Files\Ultimaker Cura 4.11.0\plugins\SimulationView\layers3d.shader`:
https://github.com/Ultimaker/Cura/blob/4.11/plugins/SimulationView/layers3d.shader#L112

Change `float t = v < 0 ? 0 : v;` to `float t = v < 0.0 ? 0.0 : v;`

bigRedBlue, 2021-11-18:
voila!
That fixed it in both 4.11 and 4.12.

I tested it on several models of varying complexity. All functionality is restored.

Thanks so much for your help!

Flow view is pretty dang neat.

closed 2021-11-18 by bigRedBlue

fieldOfView, 2021-11-18:
Thanks for testing. I have created a pull request so this fix can be included in Cura by default: https://github.com/Ultimaker/Cura/pull/10870

rburema, 2021-11-23:
This has been merged and will be available when we release the next main-line version :-)

Ressless4, 2021-11-26:
Hi I have "almost" the same issue / problem but the fix doesnt work for me. 

`Traceback (most recent call last):
  File "C:\Users\ressl\AppData\Roaming\cura\4.9\plugins\CustomJobPrefix\CustomJobPrefix\PrintInformationPatches.py", line 120, in _triggerJobNameUpdate
    self._updateJobName()
  File "C:\Users\ressl\AppData\Roaming\cura\4.9\plugins\CustomJobPrefix\CustomJobPrefix\PrintInformationPatches.py", line 129, in _updateJobName
    base_name = self._print_information._stripAccents(self._print_information._base_name)
AttributeError: 'PrintInformation' object has no attribute '_stripAccents'
`

And also my Log
`
Thread 0x000020b4 (most recent call first):
  File "threading.py", line 306 in wait
  File "C:\Users\ressl\AppData\Roaming\cura\4.9\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 2184 in wait
  File "C:\Users\ressl\AppData\Roaming\cura\4.9\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 1496 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000044fc (most recent call first):
  File "threading.py", line 306 in wait
  File "C:\Users\ressl\AppData\Roaming\cura\4.9\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 2184 in wait
  File "C:\Users\ressl\AppData\Roaming\cura\4.9\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 1315 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00002960 (most recent call first):
  File "C:\Users\ressl\AppData\Roaming\cura\4.9\plugins\OctoPrintPlugin\OctoPrintPlugin\python-zeroconf\zeroconf\__init__.py", line 1221 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00002d98 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.9.1\plugins\USBPrinting\USBPrinterOutputDeviceManager.py", line 87 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00001ab0 (most recent call first):
  File "threading.py", line 306 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1434 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00005508 (most recent call first):
  File "threading.py", line 306 in wait
  File "threading.py", line 558 in wait
  File "C:\Program Files\Ultimaker Cura 4.9.1\plugins\UM3NetworkPrinting\src\Network\ZeroConfClient.py", line 81 in _handleOnServiceChangedRequests
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000028dc (most recent call first):
  File "threading.py", line 306 in wait
  File "zeroconf\__init__.py", line 2117 in wait
  File "zeroconf\__init__.py", line 1262 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x000047a8 (most recent call first):
  File "zeroconf\__init__.py", line 1183 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

Thread 0x00003804 (most recent call first):
  File "C:\Program Files\Ultimaker Cura 4.9.1\plugins\RemovableDriveOutputDevice\RemovableDrivePlugin.py", line 61 in _updateThread
  File "threading.py", line 870 in run
  File "threading.py", line 932 in _bootstrap_inner
  File "threading.py", line 890 in _bootstrap

[... 122 lines cut ...]
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 469 in event
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1085 in event
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 452 in _show
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 469 in event
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1085 in event
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 452 in _show
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\Event.py", line 218 in call
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 469 in event
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 1085 in event
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\CrashHandler.py", line 452 in _show
[... cut ...]
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\UM\Qt\QtApplication.py", line 383 in exec_
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\lib\python3.8\site-packages\cura\CuraApplication.py", line 862 in run
  File "C:\Users\ultimaker\AppData\Local\Temp\cura-build-4.9-exe\build\inst\bin\cura_app.py", line 241 in <module>
  File "Console.py", line 36 in run
  File "__startup__.py", line 74 in run
`

My old  4.9.V Version worked around March but now it doesnt. What I did was updating my grapohics driver.

Can someone help me please?

PR #10870 (2021-11-23): Fix shader compilation on some GPUs
This PR fixes a compilation error of the shader used to render the simulation view. Some GPUs (rightly) complain about the difference between a float and an int. 

Fixes #10837
commit c6b6af66 (2021-11-23): Fix shader compilation on some GPUs

Some GPUs (rightly) complain about the difference between a float and an int. Fixes #10837