Ultimaker/Cura #10838: Must have "extra skin wall count", Model will not be slices good without any

DragonFlyer007, 2021-11-15:
### Application Version

4.11.0

### Platform

Win 10

### Printer

Ender 5 Plus

### Reproduction steps

Tried to move model, re slice, change seam, +++

### Actual results

Very strange sliced layer before top layers. (nr.45)

Found that adding at least 1 "Extra skin wall count" would fix the problem for the faulty layer (45)

### Expected results

Proper sliced model

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[Extra skin wall error.zip](https://github.com/Ultimaker/Cura/files/7541834/Extra.skin.wall.error.zip)

GregValiant, 2021-11-18:
A slice is a section through the model.  Layer 45 slices through the radius on the top edge of the part.  That fill you see on layer 45 is Cura's attempt to keep your wall thickness at .8mm as it transitions through the radius to the top skins.
If you make the model 3 walls thick that fill will go away as Cura wouldn't see the need for the extra material to satisfy the wall thickness.  With the wall count at 1 that fill spreads to more layers.

DragonFlyer007, 2021-11-18:
> A slice is a section through the model. Layer 45 slices through the radius on the top edge of the part. That fill you see on layer 45 is Cura's attempt to keep your wall thickness at .8mm as it [...]

Of course i know what a slice is.

I sliced with two walls, and it should work indipendent of how many walls i think.

Ghostkeeper, 2021-11-23:
So what's the problem with setting the Extra Skin Wall Count setting to 1, then? Your custom profile changes the setting to 0. The default profiles have it set to 1.

I'm not really seeing anything strange in the output here. It's producing a rim of skin there, yes. Maybe you want to increase the Skin Removal Width setting? It's there to remove small strips of skin where they don't really add any value. Setting that to 1.0mm (up from 0.8) removes the rim.

Let me know if that is the sort of solution you're looking for.

DragonFlyer007, 2021-11-23:
> So what's the problem with setting the Extra Skin Wall Count setting to 1, then? Your custom profile changes the setting to 0. The default profiles have it set to 1.
>
> I'm not really seeing anything [...]

Lots of people, including me likes to drag the skin out towards the outer perimeter for a cleaner top surface. Specialy on logo's and such.

So you find this behavior normal? Hmm. I dont.

Ghostkeeper, 2021-11-25:
Did the skin removal width work for you though?

And if not, could you maybe show the sort of output you're looking for, to make sure we understand the problem correctly?

closed 2021-12-09 by no-response[bot]