Ultimaker/Cura #10840: Layer area based fan % parts cooling

codecranker, 2021-11-16:
### Is your feature request related to a problem?

When my printer is working on parts with vertical delineations of small complex sections and large simple ones, it melts the small complex sections or severely over-cools the large simple ones. Min layer time is useless for controlling fan speed because the large simple sections print at much faster travel speed. For these large sections the layer time is the same even though they need like 10X less cooling because printed area is much larger.

This frequently happens printing electronics cases doing risers for PCB mounts. I either over-cool the flat base of the risers making PETG brittle and weak, or melt the pegs. I can't find a min layer time setting that works for both because printer travels so much faster when doing the flat base.

The only solution I have found is slow down travel speed 3-4X which evens out cooling requirements between layers, but more than doubles print time :(

### Describe the solution you'd like

I think there should be an option to base fan cooling % on layer cross section instead of layer times. And I think this will work so much better that it will become the default. The cooling needed for a layer depends on area of filament laid down more than anything else. And doing it this way would avoid problems I ran into with simple vs complex sections taking vastly different layer times for the same amount of filament printed.

The current solution falls over when you have parts with a mix of large and small sections that differ greatly in average travel speed. Min layer time is a bad proxy for amount of cooling needed IMO. I think printed layer area will work much better for most cases.

### Describe alternatives you've considered

Cura could analyze the size of connected meshes by layer and increase cooling for areas of layer with thin cross section. This sounds way more complicated and I'm not convinced it would work better than simply basing layer cooling % on cross sectional area of the current layer

### Affected users and/or printers

This could make parts cooling config simpler and less finicky for everyone. 

I tried manually inserting G-code for a few prints to roughly follow layer area for cooling, % and I was able to print the troublesome part at more than double the speed without melting or over-cooling.

fvrmr, 2021-11-19:
Hi @codecranker thank you for your feature request. 
We have an open feature request for fan speed for overhang. See https://github.com/Ultimaker/Cura/issues/5140
I think this is related to each other. Let me know what you think!

Ghostkeeper, 2021-11-24:
The feature request here has nothing to do with overhang. There probably are no overhangs in the PCB mounts being printed there.

I think Cura offers a lot of control over the fan speed already. There's a diagram about it in the Settings Guide:
[image]

By your story, it sounds like you need the Regular Fan Speed to be really low and the Maximum Fan Speed higher.

However the key of the feature request here is that the cooling is not just based on time but also on the area of the cooling surface. This is sort of true, but I believe that time is still by far the most important factor in how much a surface cools naturally. Thermodynamically, there are 3 important factors that determine the temperature of a printed object after printing:

* The difference in temperature between the printed plastic and the environment. The hotter the plastic or the cooler the environment, the faster it will cool down. Fans serve to make the environment cooler by replacing the air that absorbed heat from the plastic with new, room temperature air. Cura doesn't adjust its cooling based on temperatures; the profiles are supposed to factor that into its fan speed settings.
* The amount of time since printing. Heat loss is a gradual process which takes time, so the more time has passed, the cooler the plastic will become. Cura currently adjusts the fan speed based on the time. So that means that if you have something that prints very slowly (e.g. due to accelerations in skin) there is more time for the layer to cool and it won't need to turn up the fan as much.
* The contact area between the plastic and the environment. The greater the contact area, the more heat exchange can happen. This is what I think you mean to use as an argument as to why the printed area is a better metric to use.

However with the contact area metric, there is a problem. If you print faster, not only is there more surface area, but also more volume to cool down, and that takes longer. The surface area vs. volume is not linear so you could still argue that the surface area has an effect on cooling (if the sides of the lines are taken into account). But arguably, time is a much more important factor than surface area.

It could also be made to depend on both, but I think the formula would become too complex to understand for a user then. It's useful to be able to somewhat predict what a printer is going to do if you change a setting.

closed 2021-12-03 by no-response[bot]