Ultimaker/Cura #10842: Only Slice when i press ENTER

PeterSoenberg, 2021-11-16:
### Is your feature request related to a problem?

I have a slow computer so i have auto slice disabled..... it would be nice if Cura sliced when i press´´ ENTER´´ in auto slice disabled mode´´ , then you can change one setting at a time and then slice, without moving the mouse. 

### Describe the solution you'd like

Speed for slow computers

### Describe alternatives you've considered

non

### Affected users and/or printers

time

fieldOfView, 2021-11-16:
Instead of Enter you can press Ctrl + P (or Command + P on Mac OS).

fvrmr, 2021-11-19:
Hi @PeterSoenberg thank you for your request.
Because this is indeed possible with Ctrl/CMD + P I will close this issue. Since we don't plan to add or change this to enter.
I hope you understand!

closed 2021-11-19 by fvrmr