Ultimaker/Cura #10844: cura 4.12 Bounce

yomyong2, 2021-11-16:
### Application Version

4.12

### Platform

windows 10

### Printer

custom fff

### Reproduction steps

When I go into "cura" and bring in the stl file, "cura" bounces right away.

### Actual results

Cura bounce.

### Expected results

Even if I upload all the other files, it won't budge.

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

There's no file to upload

vipertech17, 2021-11-17:
mine does the same. program shuts down when trying to import an stl file. weather i drag and drop or use the folder icon on upper left hand corner of cura.

yomyong2, 2021-11-17:
I solved that problem. When Korean is included in the cura installation path, errors come out.

fvrmr, 2021-11-18:
Hi @yomyong2 thank you for your report. Good to know that you already solved it. See also this issue: https://github.com/Ultimaker/Cura/issues/9644

@vipertech17 is this also the case for you? Or do you have Chitubox installed? See otherwise here: https://github.com/Ultimaker/Cura/issues/9668

I will close this issue since it is a duplicate.

closed 2021-11-18 by fvrmr