Ultimaker/Cura #10845: Problem on Linux version

EnzoS1469, 2021-11-16:
### Application Version

4.12.0

### Platform

Ubuntu 20.04

### Printer

Flying Bear Ghost5

### Reproduction steps

1. Generated G-Code under Linux and printing failed. 
2. Generated G-Code under Windows10 using same settings and same stl file and printing was ok
3. Compared two G-Code files and differences found
4. As you can see from attached picture, one of the disks was completely missed
[image]
[FBG5_polar-scanner_parts_linux.gcode.txt](https://github.com/Ultimaker/Cura/files/7547431/FBG5_polar-scanner_parts_linux.gcode.txt)
[FBG5_polar-scanner_parts_windows.gcode.txt](https://github.com/Ultimaker/Cura/files/7547437/FBG5_polar-scanner_parts_windows.gcode.txt)
[Piatto50_current.curaprofile.txt](https://github.com/Ultimaker/Cura/files/7547455/Piatto50_current.curaprofile.txt)

[polar-scanner_parts.stl.txt](https://github.com/Ultimaker/Cura/files/7547460/polar-scanner_parts.stl.txt)
.

### Actual results

With G-Code file generated under Linux version I aborted the print after 3 or 4 attempts: because I noted that one of the discs was completely missed, I decided to check with Cura under Windows10 and printing worked. Then I compared the two G-Code files and I found a lot of differences even starting from the same stl file and from the same profile file: I was expecting a perfect match between the two files.

### Expected results

Same G-Code and same printing results

### Checklist of files to include

- [ ] Log file
- [X] Project file

### Additional information & file uploads

[FBG5_polar-scanner_parts.zip](https://github.com/Ultimaker/Cura/files/7547532/FBG5_polar-scanner_parts.zip)

EnzoS1469, 2021-11-18:
> Hi @EnzoS1469 <https://github.com/EnzoS1469> thank you for your report.
> Did you use the project file both on Windows and Linux?
> Or did you use the stl file? And changed some settings?

Hello Mr. Fenne,
I used the same stl file on both platforms and using the same profile: what
I'm experiencing even with other stl files and having the same profile file
is a different G-Code that sounds a little bit strange to me.
By the way I'm using the same Cura version - the latest one 4.12.0 - and
the same 3d printer, the FlyingBear Ghost 5.

Best Regards,
Scianni Vincenzo

smartavionics, 2021-11-20:
> By the way I'm using the same Cura version - the latest one 4.12.0 - and
the same 3d printer, the FlyingBear Ghost 5.

Really ?

```
~/Downloads/3dp$ diff FBG5_polar-scanner_parts_* | grep Generate
< ;Generated with Cura_SteamEngine master
> ;Generated with Cura_SteamEngine 4.12.0
```

Ghostkeeper, 2021-11-24:
What's more, the g-code files are virtually identical.

When put through a diff, you'll find equal printing time and material usage. There is a minute difference in the actual g-code though: The master version adds 0.00071mm of filament at some point in an inner wall. This could be a difference in the floating point precision? We've seen that happen before. As a result, all of the following E values are a few nanometres different. It seems that this propagates too, because the filament value at the end of the print is about 2.5 micrometres off.

Currently the only changes between 4.12 and master are in tree support (not used here), combing (case not hit here) and concurrency (no effect on g-code). As a result we expect the g-codes to be virtually identical. And that's also what we see here.

I think the print failure in Linux was just random variance in the printer. Perhaps the build plate wasn't as clean when the print was started from Linux. Or the filament there had a slightly different diameter. Since the g-code is virtually identical, I don't think it's a software issue.

closed 2021-11-24 by Ghostkeeper