Ultimaker/Cura #10846: Slicing fails with very complex models

fpzk, 2021-11-16:
### Application Version

4.11.0

### Platform

Windows 10 

### Printer

Geetech Rostock 

### Reproduction steps

My laptop PC is a Asus TUF A15 with Ryzen 5900H and RTX 3070 graphic card, and 16Go Ram, so an average good platform for slicing.
I've been using Cura for more than 5 years without problems and usually everything works fine.
But when I load complex models like this one : 
https://www.thingiverse.com/thing:956649/files
the slicing operation starts, and works up to 55%  then the memory usage suddently exploses to fill 99% of the memory and finally the process stops, with a message "Slicing failed"
I tried to reduce the layer number and some options, but the complexity of the model seems to be more inresponsible of the failure than the chosen options.

### Actual results

Slicing failure at 55% of the progress bar, with a "slicing failed error" message

### Expected results

Would like to slice

### Checklist of files to include

- [X] Log file
- [X] Project file

### Additional information & file uploads

[BugReport_Slice_Error_Documents.zip](https://github.com/Ultimaker/Cura/files/7547832/BugReport_Slice_Error_Documents.zip)

GregValiant, 2021-11-18:
Yes, using 4.11 that brings my laptop to it's knees also.
I had no problem when slicing with 4.12 even with the Tree Support turned on.  It took 3m15s, but it did slice.
[image]

fvrmr, 2021-11-19:
Hi @fpzk thank you for your report.
Can you check if it works for you better in 4.12?

closed 2021-12-03 by no-response[bot]